balena-io/etcher #3540: update version

franklinchang7, 2021-06-29:
- **Etcher version:5.12.0 
- **Operating system and architecture:Big sur 14
- **Image flashed:microsoft windows 10 ISO
- **Do you see any meaningful error information in the DevTools?
- 
This is the IOS with boot, but the IOS without boot appears in the program!
This program needs to be modified! Used to be normal
Thank you very much

lurch, 2021-06-29:
> Image flashed:microsoft windows 10 ISO

See #210 and https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

franklinchang7, 2021-06-29:
Dear wilson
My system is big sur 11.4
I think Etcher not support bootable write !
please check!.

thundron, 2021-07-05:
Correct, etcher does not make images bootable - you'll have to do so by yourself depending on the image, unless it's a hybrid ISO (like ubuntu for example)

closed 2021-07-05 by thundron