balena-io/etcher #3541: Upon choosing an image, Etcher goes blank

ElotonSotilas, 2021-06-29:
- **Etcher version:** 1.5.120
- **Operating system and architecture:** macOS 11.4 Big Sur
- **Image flashed:** Armbian Buster for SOPINE64
- **Do you see any meaningful error information in the DevTools?** "DevTools was disconnected from the page. Once the page is reloaded, DevTools will automatically reconnect."

I am on an M1 mac. I picked my image and instead of being allowed to pick a drive, I was met with Etcher going snow white and no content in the window. It just soft-crashed.

lurch, 2021-06-29:
Duplicate of #3520 ?

alexclaydon, 2021-07-24:
I'm experiencing this on an M1 Mac.  Reverting to v1.5.114 (an Intel build) resolved the issue.

thundron, 2021-08-17:
Please refer to https://github.com/balena-io/etcher/issues/3520 for further updates

closed 2021-08-17 by thundron