balena-io/etcher #3546: read-only my USB flash disk after flashing

Hanzhizheng, 2021-07-03:
- **Etcher version:1.5.116
- **Operating system and architecture:win10 20h1
- **Image flashed:mint
- **Do you see any meaningful error information in the DevTools?no

Hi, I have a question, after flashing my USB flash disk successfully, it turns to read-only, and can't select in etcher(display locked), I can still use it to install flashed operate system, but I can't use it to do anything else, is it broken?

lurch, 2021-07-03:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

thundron, 2021-07-05:
I highly recommend a read of https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/ ; as for the question, no it's not broken, it could be an issue with your card reader though

closed 2021-07-05 by thundron