balena-io/etcher #3548: White window, cant open etcher

peterge1998, 2021-07-05:
- **Etcher version:** 
- Using latest balena-etcher-2:1.5.120-1 from AUR
- **Operating system and architecture:** 
- Arch
- **Image flashed:** 
- nothing
- **Do you see any meaningful error information in the DevTools?** 
- Cant open etcher

running  it manually gives this:
```
➜  ~ balena-etcher-electron 
ready-to-show: 560.042ms
```
[image]

lurch, 2021-07-05:
> Using latest balena-etcher-2:1.5.120-1 from AUR

Duplicate of #3536 ?

peterge1998, 2021-07-05:
> > Using latest balena-etcher-2:1.5.120-1 from AUR
> 
> Duplicate of #3536 ?

Yes, didnt see that issue. It doesnt show the error when i try running it, but i will close my issue.

closed 2021-07-05 by peterge1998