balena-io/etcher #3552: Building on FreeBSD

grahamperrin, 2021-07-10:
```text
% pwd
/usr/home/grahamperrin/dev/etcher-1.5.120
% make
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 27: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 40: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 70: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 72: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 73: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 74: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 75: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 76: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 77: Invalid line type
make: Fatal errors encountered -- cannot continue
make: stopped in /usr/home/grahamperrin/dev/etcher-1.5.120
% 
```

Please, what to do?

lurch, 2021-07-10:
> make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 27: Invalid line type

Perhaps the syntax used at https://github.com/balena-io/etcher/blob/master/Makefile#L27 is a GNU-make extension? :shrug:

grahamperrin, 2021-07-10:
I saw _npm_ in the file and made a guess, installed [www/npm](https://www.freshports.org/www/npm/).

```text
% pkg info -x electron
electron12-12.0.9_1
% pwd
/usr/home/grahamperrin/dev/etcher-1.5.120
% npm start

> balena-etcher@1.5.120 start /usr/home/grahamperrin/dev/etcher-1.5.120
> electron .

sh: electron: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! balena-etcher@1.5.120 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the balena-etcher@1.5.120 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/grahamperrin/.npm/_logs/2021-07-10T19_29_57_492Z-debug.log
% 
```

[2021-07-10T19_29_57_492Z-debug.log](https://github.com/balena-io/etcher/files/6795832/2021-07-10T19_29_57_492Z-debug.log)

The log comprises: 

```text
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.14.8
3 info using node@v16.2.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle balena-etcher@1.5.120~prestart: balena-etcher@1.5.120
6 info lifecycle balena-etcher@1.5.120~start: balena-etcher@1.5.120
7 verbose lifecycle balena-etcher@1.5.120~start: unsafe-perm in lifecycle true
8 verbose lifecycle balena-etcher@1.5.120~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/usr/home/grahamperrin/dev/etcher-1.5.120/node_modules/.bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin:/home/grahamperrin/bin:/home/grahamperrin/.local/bin
9 verbose lifecycle balena-etcher@1.5.120~start: CWD: /usr/home/grahamperrin/dev/etcher-1.5.120
10 silly lifecycle balena-etcher@1.5.120~start: Args: [ '-c', 'electron .' ]
11 info lifecycle balena-etcher@1.5.120~start: Failed to exec start script
12 verbose stack Error: balena-etcher@1.5.120 start: `electron .`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
12 verbose stack     at ChildProcess.emit (node:events:365:28)
12 verbose stack     at maybeClose (node:internal/child_process:1067:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
13 verbose pkgid balena-etcher@1.5.120
14 verbose cwd /usr/home/grahamperrin/dev/etcher-1.5.120
15 verbose FreeBSD 14.0-CURRENT
16 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
17 verbose node v16.2.0
18 verbose npm  v6.14.8
19 error code ELIFECYCLE
20 error syscall spawn
21 error file sh
22 error errno ENOENT
23 error balena-etcher@1.5.120 start: `electron .`
23 error spawn ENOENT
24 error Failed at the balena-etcher@1.5.120 start script.
24 error This is probably not a problem with npm. There is likely additional logging output above.
25 verbose exit [ 1, true ]
```

grahamperrin, 2021-07-10:
With [www/npm-node14](https://www.freshports.org/www/npm-node14/) in lieu of www/npm (I don't know the difference: 

```text
% pwd
/usr/home/grahamperrin/dev/etcher-1.5.120
% npm start

> balena-etcher@1.5.120 start /usr/home/grahamperrin/dev/etcher-1.5.120
> electron .

sh: electron: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! balena-etcher@1.5.120 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the balena-etcher@1.5.120 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/grahamperrin/.npm/_logs/2021-07-10T19_36_29_300Z-debug.log
% 
```

grahamperrin, 2021-07-10:
> `… did you mean to install? …`

Well, yes, but npm is a novelty to me, so I took another guess. 

`npm install` failed: 

[2021-07-10T19_41_21_164Z-debug.log](https://github.com/balena-io/etcher/files/6795847/2021-07-10T19_41_21_164Z-debug.log)

From the log: 

```text
42153 verbose stack Error: @balena.io/usb@1.3.14 install: `node-pre-gyp install --fallback-to-build`
42153 verbose stack Exit status 1
42153 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
42153 verbose stack     at EventEmitter.emit (events.js:376:20)
42153 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
42153 verbose stack     at ChildProcess.emit (events.js:376:20)
42153 verbose stack     at maybeClose (internal/child_process.js:1055:16)
42153 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
42154 verbose pkgid @balena.io/usb@1.3.14
42155 verbose cwd /usr/home/grahamperrin/dev/etcher-1.5.120
42156 verbose FreeBSD 14.0-CURRENT
42157 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
42158 verbose node v14.17.0
42159 verbose npm  v6.14.8
42160 error code ELIFECYCLE
42161 error errno 1
42162 error @balena.io/usb@1.3.14 install: `node-pre-gyp install --fallback-to-build`
42162 error Exit status 1
42163 error Failed at the @balena.io/usb@1.3.14 install script.
42163 error This is probably not a problem with npm. There is likely additional logging output above.
42164 verbose exit [ 1, true ]
```

lurch, 2021-07-10:
> I took another guess. `npm install` failed:

https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md#installing-npm-dependencies

grahamperrin, 2021-07-11:
```text
% pwd
/usr/home/grahamperrin/dev/etcher-1.5.120
% make electron-develop
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 27: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 40: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 70: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 72: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 73: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 74: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 75: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 76: Invalid line type
make: "/usr/home/grahamperrin/dev/etcher-1.5.120/Makefile" line 77: Invalid line type
make: Fatal errors encountered -- cannot continue
make: stopped in /usr/home/grahamperrin/dev/etcher-1.5.120
% gmake electron-develop
gmake: /bin/bash: No such file or directory
gmake: /bin/bash: No such file or directory
Makefile:73: *** We could not detect your host platform.  Stop.
% which bash
/usr/local/bin/bash
% 
````

grahamperrin, 2021-07-11:
### Available versions of electron

```text
% pkg search electron
electron12-12.0.9_1            Build cross-platform desktop apps with JavaScript, HTML, and CSS
% 
```

[devel/electron12](https://www.freshports.org/devel/electron12/)

https://github.com/electron/electron/issues/3797#issuecomment-845428987

closed 2023-07-12 by dfunckt