balena-io/etcher #3554: Security Issue Update

cloeg97, 2021-07-13:
Hello,

I have been looking into using Balena’s Etcher instead of Win32 Disk Imager that I currently use. However, during my research I have some across numerous of issues that were flagged around security. I was wondering if I could get an update on the [actions mentioned ](https://github.com/balena-io/etcher/issues/2766#issuecomment-531437542) in issue #2766.

I found this to be the most informative about the facts of the issue and where progress was with it. However, this was back in September 2019. I have had a look around for an update on these actions mentioned in this comment but have been unsuccessful. I would like to us this software, but I wanted to check the progress of the mentioned actions to resolve the security issues. 
 
Would anyone be able to provide me with an update on this?

thundron, 2021-07-13:
Hi, well we fixed the issue shortly after it happened, we just needed to gather information on our end on what services we were making connections to and normalize them + remove any trace of PIID since our intention has always been to just collect anonymous data, so it was just a matter of fixing an issue rather than changing the behaviour/scope of our analytics and as I mentioned [in that issue](https://github.com/balena-io/etcher/issues/2766#issuecomment-553396158) this was done pretty much without the need to release a new version if not for a couple parameters/configurations that needed adjustment.
I guess some information was lost in the midst of comments as the conversation hasn't been as healthy as other

cloeg97, 2021-07-13:
@thundron that is great news. Thank you for the clarification on this.

I am happy to hear this has all been cleared up.

closed 2021-07-13 by cloeg97