balena-io/etcher #3557: Etcher fails to install on openSUSE Tumbleweed 20210711 x86-64

jdrch, 2021-07-14:
- **Etcher version:** v1.5.121
- **Operating system and architecture:** openSUSE Tumbleweed 20210711 x86-64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

Etcher fails to install on openSUSE Tumbleweed 20210711 x86-64. Here's what I get when I try to do so:

```
$ curl -1sLf 'https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh' | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'rpm' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=opensuse-tumbleweed  version=20210711  codename=<empty>  arch=x86_64
 >>>>:
   OK: Importing 'balena/etcher' repository GPG keys into rpm ...
   OK: Checking for available package manager (DNF/YUM/Zypper) ...
 ^^^^: ... Detected package manager as 'zypper'
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=opensuse-tumbleweed  version=20210711  codename=<empty>  arch=x86_64
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as SUSE Linux Enterprise Server 12, use the following:
 >>>>:
 >>>>:   <curl command> | distro=sles version=12.1 codename= sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!:
 !!!!: You can contact balena for further assistance.
 !!!!:
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!:
 !!!!: Last 20 log lines from .csm_setup_RCb01i3PB (might not be errors, nor even relevant):
 !!!!:
```

Any ideas?

tsosrmr, 2021-09-12:
I got it to install with

`curl -1sLf   'https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh'   | codename="tumbleweed" sudo -E bash`

followed by 

`sudo zypper in balena-etcher-electron`

Though there was a warning during the install:

```
Additional rpm output:
chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory
```

What I don't get is why I need a codename for a distro that doesn't use them.

jdrch, 2021-09-13:
Thanks, that worked!

> What I don't get is why I need a codename for a distro that doesn't use them.

I think that's for the repo, not the distro. I'm guessing different builds are made depending on the distro.

1fallen, 2022-11-04:
I can load the repo with same warning "jdrch" shows and it installs but fails to launch:
` /opt/balenaEtcher/balena-etcher-electron %U

(balena-etcher-electron.bin:16178): Gtk-WARNING **: 08:43:02.326: Theme parsing error: gtk.css:4091:13: negative values are not allowed.

(balena-etcher-electron.bin:16178): Gtk-WARNING **: 08:43:02.333: Theme parsing error: gtk.css:7855:13: negative values are not allowed.

(balena-etcher-electron.bin:16178): Gtk-WARNING **: 08:43:02.333: Theme parsing error: gtk.css:8039:19: negative values are not allowed.

(balena-etcher-electron.bin:16178): Gtk-WARNING **: 08:43:02.334: Theme parsing error: gtk.css:8317:21: negative values are not allowed.

(balena-etcher-electron.bin:16178): Gtk-WARNING **: 08:43:02.334: Theme parsing error: gtk.css:8318:24: negative values are not allowed.
[Axios v1.7.7] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.7] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.7] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[16178:1104/084302.783358:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/opt/balenaEtcher/balena-etcher-electron: line 10: 16178 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"
:`

closed 2023-07-12 by dfunckt