balena-io/etcher #3558: White screen on Balena Etcher

helpihavenoeyebrows, 2021-07-14:
- Balena-Etcher-2:1.5.120-1
- Arch Linux, amd64 

I've updated to the newest edition of Balena Etcher and when doing so, all I get is a white screen for the application and nothing else. When opening Devtools and looking at the application, it looks like this and.
While I may be no HTML expert, I know this is off.

Can anyone please help?
[image]
[image]

thundron, 2021-07-15:
can you check if there's any error in the Console tab of the DevTools ?

lurch, 2021-07-15:
Similar to #2866 (amongst others) ?

alexclaydon, 2021-07-24:
I'm experiencing this on an M1 Mac.  Reverting to v1.5.114 (an Intel build) resolved the issue.

lilithium-hydride, 2021-07-26:
Getting this on Artix with aur/balena-etcher 2:1.5.120-1. @thundron OP seems to have disappeared, but mine is exhibiting the same behavior with the following in the console:
```
index.css:1 Failed to load resource: net::ERR_FILE_NOT_FOUND
```
```
electron/js2c/asar_bundle.js:5 Uncaught Error: /usr/lib/balena-etcher/generated/modules/xxhash/build/Release/hash.node: undefined symbol: _ZN2v86Object18InternalFieldCountEv
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5)
    at Module.load (internal/modules/cjs/loader.js:935)
    at Module._load (internal/modules/cjs/loader.js:776)
    at Function.f._load (electron/js2c/asar_bundle.js:5)
    at Function.o._load (electron/js2c/renderer_init.js:33)
    at Module.require (internal/modules/cjs/loader.js:959)
    at require (internal/modules/cjs/helpers.js:88)
    at Object.defineProperty.value (gui.js:4216)
```
```
electron/js2c/renderer_init.js:113 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
    Policy set or a policy with "unsafe-eval" enabled. This exposes users of
    this app to unnecessary security risks.

For more information and help, consult
https://electronjs.org/docs/tutorial/security.
This warning will not show up
once the app is packaged.
(anonymous) @ electron/js2c/renderer_init.js:113
```

thundron, 2021-08-17:
See https://github.com/balena-io/etcher/issues/3536 for arch

thundron, 2021-08-17:
thanks @lilithium-hydride for the extra context!

closed 2021-08-17 by thundron