balena-io/etcher #3559: Windows Flash Failed Error

cloeg97, 2021-07-15:
- **Etcher version:** 1.5.120
- **Operating system and architecture:** Windows 10 Pro, Intel(R) Core(TM) i7-10850H CPU @2.70GHz 64-bit
- **Image flashed:** linux image
- **Do you see any meaningful error information in the DevTools?** See below for snippit

Done` ({"image":{"size":1210113024,"compressedSize":247822197,"hasMBR":true,"path":"C:\\Users\\.............\\.............\\Documents\\core-image-minimal-dev-es-20210428094631.rootfs.wic.gz","partitions":[{"offset":4194304,"size":19077120,"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","guid":"BF599678-49A5-4072-A690-06851E5E86F8","name":"bootry","index":1},{"offset":23271424,"size":8388608,"type":"21686148-6449-6E6F-744E-656564454649","guid":"15211F8C-06ED-42F6-BF81-9A59AFA959EE","name":"primary","index":2},{"offset":33554432,"size":1176541184,"type":"0FC63DAF-8483-4772-8E79-3D69D8477DE4","guid":"74ED0CB0-881A-46E2-BEB8-CA0D67397D08","name":"rootry","index":3}],"isCompressed":true,"name":"core-image-minimal-dev-es-20210428094631.rootfs.wic","isSizeEstimated":true,"extension":"gz"},"drives":[{"size":7820083200,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\, , "}],"description":"MSCC PolarFireSoC_msd USB Device","isUAS":false,"partitionTableType":"gpt"}],"driveCount":1,"uuid":"543c7581-0c63-4bab-9af8-6a7bbb420e01","status":"finished","flashInstanceUuid":"543c7581-0c63-4bab-9af8-6a7bbb420e01","errors":[{"name":"Error","description":"MSCC PolarFireSoC_msd USB `Device","device":"\\\\.\\PhysicalDrive1"}],"devices":{"failed":1,"successful":0},"bytesWritten":1001446400,"sourceMetadata":{"size":1210113024,"name":"decompressed-image-a5992794993d.tmp","blocks":[{"blocks":[{"offset":0,"length":1001390080}]},{"blocks":[{"offset":1210056704,"length":56320}]}],"blockmappedSize":1001446400},"sample":0.1,"applicationSessionUuid":"9f509b30-fedb-454a-ba2d-25d920104491","flashingWorkflowUuid":"1d453699-5607-4259-b778-993db7890ed3"})

lurch, 2021-07-15:
"MSCC PolarFireSoC_msd USB Device" looks unusual - I wonder if it might work with a different USB device / card-reader?

cloeg97, 2021-07-15:
@lurch Thank you for your response. However, it is not the device that is the issue as the same works perfectly when using other disk image software's. The device is an FPGA dev board which is why the name may look unfamiliar.

CaelThunderwing, 2021-07-20:
im getting a Similar Issue since the Last Cumulative Update in Windows 10,

Tue Jul 20 2021 01:18:22 GMT-0400 (Eastern Daylight Time) Flash error ({"image":{"size":39648755712,"name":"PIMGA2BETA2.img","hasMBR":true,"partitions":[{"offset":4194304,"size":268435456,"type":12,"index":1},{"offset":384827392,"size":39263928320,"type":131,"index":2},{"offset":272629760,"size":112197632,"type":6,"index":3}],"extension":"img","path":"D:\\PIMGA2BETA2.img"},"drives":[{"size":63864569856,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive4","error":null,"isReadOnly":false,"displayName":"I:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive4","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"I:\\"}],"description":"SD Card Reader USB Device","isUAS":false,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"d40058c5-1e6e-4bfd-b522-db7c5efe50d2","flashInstanceUuid":"d40058c5-1e6e-4bfd-b522-db7c5efe50d2","error":{"name":"Error","message":"EPERM: operation not permitted, write","stack":"Error: EPERM: operation not permitted, write","code":"EPERM","syscall":"write","errno":-4048},"sample":0.1,"applicationSessionUuid":"5e85f481-862f-45a4-a3b4-5984b2600158","flashingWorkflowUuid":"72434d52-050c-480b-957c-d5614c30b7a8"})

though balena etcher isnt the only thing apparently affected,  dotNet Disk Imager too gets a Error on getting Write access, yet the RaspberryPi Imager gets access.

closed 2023-07-12 by dfunckt