balena-io/etcher #3560: 1.5.121 - sha512 mismatch on mac zip YAML assets

luckman212, 2021-07-15:
1. from https://github.com/balena-io/etcher/releases/download/v1.5.121/latest-mac.yml :

```yaml
version: 1.5.121
files:
  - url: balenaEtcher-1.5.121.zip
    sha512: AsyztoApCdRN1gp8XUBKj6mhHJQOjqWDPDJAtOlv+PLzDYU1dBUrVobR5j0g/aTd2FRUZEBq68wqd4eDV77dNA==
    size: 178506211
    blockMapSize: 173576
path: balenaEtcher-1.5.121.zip
sha512: AsyztoApCdRN1gp8XUBKj6mhHJQOjqWDPDJAtOlv+PLzDYU1dBUrVobR5j0g/aTd2FRUZEBq68wqd4eDV77dNA==
releaseDate: 2021-07-05T18:42:54.555Z
stagingPercentage: 0
```

so, `sha512: AsyztoApCdRN1gp8XUBKj6mhHJQOjqWDPDJAtOlv+PLzDYU1dBUrVobR5j0g/aTd2FRUZEBq68wqd4eDV77dNA==`

2. download [balenaEtcher-1.5.121.zip](https://github.com/balena-io/etcher/releases/download/v1.5.121/balenaEtcher-1.5.121.zip)

3. compute hash:
```shell
$ openssl dgst -sha512 -binary "balenaEtcher-1.5.121.zip" | base64
IGcRunUaQLnIO52qw5qzuxPHpwWL7Rtn4E812PH4TEpFqsqmlO/6EF9q25z0rcOX+4Vu/H3rEJMsRvFoNgKtUw==
```

lurch, 2021-07-15:
See https://github.com/balena-io/etcher/issues/2826#issuecomment-500174949

luckman212, 2021-07-15:
@lurch I read the linked issue but still not sure how that relates. I believe I am running the right command to decode the sha512sum, it's just that the checksum in the yaml does not match the computed output of the bits.

lurch, 2021-07-15:
I dunno - I don't work on Etcher so I've no idea (and personally don't care) whether the checksums match or not - I was just providing a link to a related issue which has helped in the past when other people have complained about the checksums not matching :slightly_smiling_face:  :shrug: 
Hopefully @thundron will pop along at some point and take a look?