balena-io/etcher #3563: "Oh no, your setup failed"

dark0ghost, 2021-07-20:
OS: Parrot os home
Version os: 4.11
Kernel: 5.10.0-6parrot1-amd64

os is based on Debian(10 buster), but when I activate ```curl -1sLf  'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh`   | sudo -E bash``` i get error: *.. Failed to update via apt-get update - Context above. !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)*
__full__:
```bash
curl -1sLf \
   'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
   | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=parrot  version=4.11  codename=4.11  arch=x86_64  
 >>>>:
 NOPE: Checking for apt dependency 'apt-transport-https' ...
 NOPE: Updating apt repository metadata cache ...
Failed to update via apt-get update

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Сущ:1 http://dl.google.com/linux/chrome/deb stable InRelease
Сущ:3 https://download.docker.com/linux/debian buster InRelease
Сущ:2 https://apt.llvm.org/buster llvm-toolchain-buster InRelease
Игн:5 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease
Пол:6 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Release [697 B]
Сущ:7 https://deb.parrot.sh/parrot rolling InRelease
Пол:8 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Release.gpg [836 B]
Сущ:9 https://deb.parrot.sh/parrot rolling-security InRelease
Игн:8 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Release.gpg
Ошб:4 https://dl.bintray.com/etcher/debian stable InRelease
  403  Forbidden [IP: 34.211.43.146 443]
Чтение списков пакетов…
W: Цель Packages (main/binary-amd64/Packages) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Packages (main/binary-all/Packages) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Translations (main/i18n/Translation-ru_RU) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Translations (main/i18n/Translation-ru) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Translations (main/i18n/Translation-en) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Ошибка GPG: http://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Release: Следующие подписи не могут быть проверены, так как недоступен открытый ключ: NO_PUBKEY F60F4B3D7FA2AF80
E: Репозиторий «http://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Release» не подписан.
E: Не удалось получить https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 34.211.43.146 443]
E: Репозиторий «https://deb.etcher.io stable InRelease» не подписан.
W: Цель Packages (main/binary-amd64/Packages) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Packages (main/binary-all/Packages) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Translations (main/i18n/Translation-ru_RU) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Translations (main/i18n/Translation-ru) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
W: Цель Translations (main/i18n/Translation-en) настроена несколько раз: в /etc/apt/sources.list.d/google-chrome.list:3 и в /etc/apt/sources.list.d/google.list:1
 ^^^^: ... Failed to update via apt-get update - Context above.
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 

```

lurch, 2021-07-20:
> Ошб:4 https://dl.bintray.com/etcher/debian stable InRelease
  403  Forbidden [IP: 34.211.43.146 443]
E: Не удалось получить https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 34.211.43.146 443]

Looks like you need to remove the old bintray repo from your sources.list, see #3501

dark0ghost, 2021-07-20:
Deleting the old repository didn't change anything. The error remained

lurch, 2021-07-20:
Well AFAIK you need to get `apt-get update` working "cleanly" (i.e. with no errors) first, before the setup script will work.

dark0ghost, 2021-07-20:
Before opening the issue I did this but nothing changed and I opened this issue

dark0ghost, 2021-07-20:
Thk. Unfortunately the script does not work with parrot os

closed 2021-07-20 by dark0ghost