balena-io/etcher #3566: Cannot write FreeDOS bootable floppy image to a real 1.44MB floppy disk from Windows

juj, 2021-07-24:
- **Etcher version:** 1.5.121
- **Operating system and architecture:**  Windows 10 Pro 64-bit
- **Image flashed:** FD13FLOP.img contained in FD13-Floppy.zip from https://www.ibiblio.org/pub/micro/pc-stuff/freedos/files/distributions/1.3/previews/1.3-rc4/
- **Do you see any meaningful error information in the DevTools?** No

I want to write FreeDOS boot disk to a real 1.44MB floppy disk from a Windows 10 host so that I can boot my 386 PC from it.

For that purpose, I have the [Freecom USB Floppy Disk Drive](https://www.freecom.com/product/usb-floppy-disk-drive-22767/) on my Windows 10 system, which recognizes as a drive A:\ on Windows:

[image]

to which I can navigate in Explorer as usual:

[image]

Etcher does not recognize that drive as a flash target however, but presents the following screen when choosing the target device/drive to write to:

[image]

Oddly the drive identifier `A:\` is misassociated to the Samsung system SSD drive, which is not correct.

Any chance this might be a regression? (Has this use case ever worked?)

juj, 2021-07-24:
Thought to try the same from a macOS host, but looks like that Freecom drive is busted there - after connecting it to macOS via USB, the drive motor keeps spinning and never settles, and attempting to access the drive gives "Resource is busy" error.

Then trying from a Linux host, I then run into issues getting Etcher working.

[image]

lurch, 2021-07-24:
> Oddly the drive identifier A:\ is misassociated to the Samsung system SSD drive, which is not correct.

How odd! Maybe that's a bug in https://github.com/balena-io-modules/drivelist ? :shrug: 

> Then trying from a Linux host, I then run into issues getting Etcher working.

See #3129

juj, 2021-07-24:
In the end, I was able to achieve what I was looking for by running `dd if=FD13FLOP.img of=/dev/sdb` on Linux, and I got a working floppy disk. This can be closed or kept tracked for future.

closed 2021-08-17 by thundron

gratejames, 2026-08-17:
I know this is an old issue, but I'm seeing this _exact_ same behavior as I try to write a Macintosh System image with a Dell USB floppy drive. Attached is a screenshot matching the one that opened the issue. The floppy drive is mounted to A:\, but misattributed to the Samsung C:\ drive.

[image]