balena-io/etcher #3574: Etcher appimage not opening in Ubuntu20.04

yousufAzadAyon, 2021-08-10:
Doubling clicking the appimage did not work so I tried the following commands 

    chmod +x .appimage
    sudo ./.appimage
but it gives me the error
<pre>dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
</pre>

but Fuse is installed in my system.
<pre>apt-cache policy libfuse2
libfuse2:
  Installed: 2.9.9-3
  Candidate: 2.9.9-3
  Version table:
 *** 2.9.9-3 500
        500 http://mirror.xeonbd.com/ubuntu-archive focal/main amd64 Packages
        100 /var/lib/dpkg/status
</pre>
I'm on Ubuntu 20.04
balenaEtcher-1.5.121-ia32

lurch, 2021-08-10:
"500 http://mirror.xeonbd.com/ubuntu-archive focal/main **amd64** Packages"
"balenaEtcher-1.5.121-**ia32**"

You probably need to use the 64-bit AppImage rather than the 32-bit AppImage?

yousufAzadAyon, 2021-08-10:
Yes, the 64-bit worked just fine. I dont know how I missed this detail. :D

closed 2021-08-10 by yousufAzadAyon

tredondo, 2025-06-07:
Where is this 64-bit appImage? I don't see it at https://etcher.balena.io/#download-etcher