balena-io/etcher #3579: problem finding /opt/balena-etcher-electron/chrome-sandbox

ArjenR49, 2021-08-18:
- **Etcher version:** 
- **Linux Mint Debbie 4, Cinnamon 32-bit ** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Installation from repository following the instructions above gives the following error message at the very end:
 chmod: kan geen toegang krijgen tot &#39;/opt/balena-etcher-electron/chrome-sandbox&#39;: Bestand of map bestaat niet
or in English approximately:
chmod: cannot gain access to  &#39;/opt/balena-etcher-electron/chrome-sandbox&#39;: file or folder doesn't exist

INdeed there is no such folder in /opt. There is a folder balenaEtcher, which contains the sandbox file. However renaming the balenaEtcher folder to balena-etcher-electron didn't solve the problem: balena etcher doesn't start.
I also uninstalled and installed again. No error message this time, but no dice, either.

thundron, 2021-08-20:
Hi, have you tried the solution suggested at https://github.com/balena-io/etcher/issues/3486#issuecomment-831861956 (comment) in the meantime ?

ArjenR49, 2021-08-20:
Yes, I found and did the trick with the symbolic link and it worked.

Thank you for asking. I am trying to keep an HP mini-notebook 5101 alive 
as a spare machine. It is old, one CPU, a 32-bit Atom, but the hardware 
is still doing fine.

Next the eternal problem with Broadcom wireless. They should be barred 
... I'll have to find a WiFi adapter that works. If only there were 
Intel WiFi adapters ... Since the mini-notebook never any other built-in 
than Intel WiFi for me ... Other brands are just too much trouble and 
uncertainty.

jferments, 2025-06-17:
I am still getting this error in Ubuntu 24.04 ...

ethagnawl, 2026-09-14:
How are these workarounds still required? I've had to stumble across these (helpful!) threads every time I've tried to use Etcher in the last ~10 years.

seb-1204, 2026-10-25:
> 6\. How are these workarounds still required? I've had to stumble across these (helpful!) threads every time I've tried to use Etcher in the last ~10 years.

100% same thought by I had, I’m seeing the same `chrome-sandbox` path mismatch with balenaEtcher 2.1.9 on Ubuntu 26.04.

During installation, the package reports:

```bash
chmod: cannot access '/opt/balenaEtcher/chrome-sandbox': No such file or directory
```

But the helper is installed here:

```bash
/usr/lib/balena-etcher/chrome-sandbox
```