balena-io/etcher #3583: Broken Cruzer Blade 8Gb

FrenchBen, 2021-08-22:
- **Etcher version: 1.5.121
- **Operating system and architecture:** MacOS 11.5.2 
- **Image flashed:** `ubuntu-20.04.2.0-desktop-amd64.iso`
- **Do you see any meaningful error information in the DevTools?** Nope

Adding myself to the list of the "lost usb" crew here - 
1. Formatted disk with DiskUtil
2. Started Etcher (recognized disk)
3. Started flashing process
4. Etcher asked for elevated permissions to use drive that was no longer mounted (accepted)
5. Etcher reported a failed flash
6. USB disk no longer found in MacOS

Tried to rescue the disk following the directions here:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

USB drive doesn't show up in any disk utilities.  :(

prorounak2005, 2021-08-23:
Same bro I use san disk and same thing happened- https://github.com/balena-io/etcher/issues/3585 
U can run this command in ur command prompt (run as administrator) - 
[image]

FrenchBen, 2021-08-23:
Thx @prorounak2005 for chiming in.
1. I'm not on windows, so I can't run the above
2. I can't even view the disk anymore, in any diskutil/diskpart tools

I'm guessing etcher killed part of the MBR? My output is very much like this:
https://gist.github.com/cristyansv/131cb4ea8d76487ebe68d31e4137515e

prorounak2005, 2021-08-23:
maybe this help - https://www.easeus.com/mac-file-recovery/usb-flash-drive-not-showing-up-on-mac.html

prorounak2005, 2021-08-23:
Etcher is trash use rufus - https://rufus.ie/en/

prorounak2005, 2021-08-23:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable