balena-io/etcher #3585: Balena corrupted my USB pendrive

prorounak2005, 2021-08-23:
- **Etcher version:**  1.5.121
- **Operating system and architecture:** Windows 10 
- **Image flashed:** Yes 
- **Do you see any meaningful error information in the DevTools?** No

Balena etcher corrupted my pen drive. Or I think so. Because It was a new pendrive and after I flashed a windows 10 in my usb pendrive and opened this pc. I was not able to find my usb.  
[image]

prorounak2005, 2021-08-23:
I am not able to see my F drive.

lurch, 2021-08-23:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives ?

prorounak2005, 2021-08-23:
> 
> 
> https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives ?

Ya I formatted my usb. Now how will I install windoows 10 in my usb ? I want to make a bootable usb.

prorounak2005, 2021-08-23:
> 
> 
> https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives ?

Whenever I try to flash windows 10 iso in my usb drive. The usb gets corrupted. What should I do ?

lurch, 2021-08-23:
> Now how will I install windoows 10 in my usb ? I want to make a bootable usb.

https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

prorounak2005, 2021-08-23:
bruh it is trash

closed 2021-08-23 by prorounak2005

beamulator, 2022-09-02:
i tried this method but it continues to state thst there is no media in the usb drive.  I was flashing batocera to usb via balena etcher and recieved an error message at 99%.  now no matter what i do I can not access the drive.  when 
 i run disk psrt and run the clean command it does not detect the usb drive.  HELP!!! this is a brand new 128gb high speed drive that just got trashed.