balena-io/etcher #3588: automatically selecting hard drive

Wjahroll, 2021-09-01:
A program like this should NEVER EVER automatically select the hard drive that gets flashed.

I was trying to flash the correct usb, i selected it and i noticed that the usb wasnt flashing so i canceled it.
Then i was taken back to the flash menu but what i didnt notice was that it had automatically selected my flash drive with all my great and valuable data in it!! As you can guess, i tried flashing it again.

Now i am very happy that I have basically lost very valuable data, i would like this "feature" to be removed.

Wjahroll, 2021-09-01:
I feel like i am going to puke

thundron, 2021-09-06:
I'm sorry that you lost some data, though that is no reason to be disrespectful.
With that out of the way, I would be interested in knowing what you selected, as the automatic selection happens only when you insert new targets and/or at the application start; it doesn't happen, for example, after you have selected another target.
If there's a bug in that, I would like to know more about it and resolve it as it can potentially impact other people.
That will require the application logs and some more information that you can find in the issue opening template.

DecanFrost, 2021-09-10:
Had the same issue, it auto selected my 3tb MOVIE drive and i lost everything.

thundron, 2021-09-13:
keep in mind that auto-selection works only for targets with less than 128GB and only if that target is the only one connected, in all other cases the target has to be manually selected