balena-io/etcher #3591: Can't select SD card reader as a target when flash file is on external drive

kirill-savchuk, 2021-09-02:
- **Etcher version:** Version = 1.5.121, Type = dmg
- **Operating system and architecture:** macOS 11.5.2 (20G95) Intel
- **Image flashed:** 2021-05-07-raspios-buster-armhf-full.img
- **Do you see any meaningful error information in the DevTools?** No errors

`Apple SD Card Reader` is marked as **Source Drive** and is inactive. And I cannot select it as a target. But in reality, everything is the opposite. My IMG file is on an external drive and I want to flash it on sd card.

[image]

Logs

```
{"size":8602517504,"name":"2021-05-07-raspios-buster-armhf-full.img","hasMBR":true,"partitions":[{"offset":4194304,"size":268435456,"type":12,"index":1},{"offset":272629760,"size":8329887744,"type":131,"index":2}],"extension":"img","path":"/Volumes/Samsung 256G/Raspberry Pi/2021-05-07-raspios-buster-armhf-full.img","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"dcfb928c-b470-4676-9ae3-a9b36f3f5dd8","flashingWorkflowUuid":"614f9de8-c250-49b3-a5e8-92bac9818095"})
```

As a workaround, I copy IMG file to a hard drive.

bartversluijs, 2021-10-05:
I'm having the same issue in version 1.5.122 and in version 1.6.0.
My image is on an external drive and my SD card is detected as the source drive, so I can't flash the SD card with the image.

dfunckt, 2023-07-12:
Closing in favor of #4051

closed 2023-07-12 by dfunckt