balena-io/etcher #3592: balenaEtcher window shows "Etcher" instead of "balenaEtcher" on windows version

danthegoodman1, 2021-09-03:
See attached screenshot, the header shows `Etcher` instead of `balenaEtcher`, should show `balenaEtcher`.

macOS already shows `balenaEtcher`, so this is isolated to Windows.

[image]

sergiotca, 2021-09-09:
I'm not sure it is "isolated" to Windows...
This is 1.5.117 running on Ubuntu 21.04:
[image]

danthegoodman1, 2021-09-09:
Good find!

sergiotca, 2021-09-09:
TBH I suspect this title is by design. I could be mistaken but I believe "Etcher" is the application's actual name.
Balena would be kind of like its ecosystem.

danthegoodman1, 2021-09-09:
On one of the Friday all hands Alex mentioned that it should be
balenaEtcher so I made the pr.

sergiotca, 2021-09-09:
Oh, I was not aware of that.
In any case, I guess naming consistency among its deployments would be desired.

lurch, 2021-09-10:
Tangentially related to #3209

There's also a bunch of places in `README.md` which say "Etcher" which should perhaps be "balenaEtcher"?
And I see `package.json` says `"author": "Balena Inc. <hello@etcher.io>",` but I dunno if you're still using the etcher.io domain?

Ahh, in fact:
```
$ git grep 'etcher\.io'|grep -v CHANGELOG
SUPPORT.md:[faq]: https://etcher.io
docs/MAINTAINERS.md:  - **E-mail address:** hello@etcher.io
lib/gui/menu.ts:							'https://etcher.io/pro?utm_source=etcher_menu&ref=etcher_menu',
lib/gui/menu.ts:						electron.shell.openExternal('https://etcher.io?ref=etcher_menu');
package.json:  "author": "Balena Inc. <hello@etcher.io>",
```
and also:
```
$ git grep 'balenaEtcher'|grep -v CHANGELOG|wc -l
10
$ git grep 'BalenaEtcher'|grep -v CHANGELOG|wc -l
3
$ git grep '\bEtcher\b'|grep -v CHANGELOG|wc -l
83
```

commit 4c5d052a (2022-03-10): patch: updated title to balenaEtcher

fixes #3592

Change-Type: patch
closed 2022-03-18 by bulldozer-balena[bot]