balena-io/etcher #3593: regression: 1.5.122 is not 'universal' for both Intel and ARM/M1/AppleSilicon Macs anymore

core-code, 2021-09-04:
1.5.121 was universal, 1.5.122 is not anymore:

```
%file /Volumes/balenaEtcher\ 1.5.121-universal/balenaEtcher.app/Contents/MacOS/balenaEtcher

/Volumes/balenaEtcher 1.5.121-universal/balenaEtcher.app/Contents/MacOS/balenaEtcher: Mach-O universal binary with 2 architectures: [x86_64:Mach-O 64-bit executable x86_64] [arm64:Mach-O 64-bit executable arm64]
/Volumes/balenaEtcher 1.5.121-universal/balenaEtcher.app/Contents/MacOS/balenaEtcher (for architecture x86_64):	Mach-O 64-bit executable x86_64
/Volumes/balenaEtcher 1.5.121-universal/balenaEtcher.app/Contents/MacOS/balenaEtcher (for architecture arm64):	Mach-O 64-bit executable arm64

% file /Volumes/balenaEtcher\ 1.5.122/balenaEtcher.app/Contents/MacOS/balenaEtcher          
/Volumes/balenaEtcher 1.5.122/balenaEtcher.app/Contents/MacOS/balenaEtcher: Mach-O 64-bit executable x86_64

```

blitzcaster, 2021-09-28:
Any news on this? I'm currently stuck at 1.5.121 on MBP M1.

thundron, 2021-09-29:
No news yet, see the other issue linked

core-code, 2021-11-11:
issue persists in 1.7.0

n0rt0nthec4t, 2021-11-16:
Add me to the issue,, when can we expect to have a universal binary back?!>?!?

danielktdoranie, 2021-11-24:
> Add me to the issue,, when can we expect to have a universal binary back?!>?!?

Same here I can confirm that 1.7.0, as downloaded from https://www.balena.io/etcher/ STILL requires Rosetta 2 to work and is NOT Apple Silicon/M1 native. 

What's the deal here? I understand we had support and then it was removed? When will it be back?

nodesocket, 2021-12-03:
Also echoing why native `Apple Silicon` _(Universal)_ support was lost? Running latest version of  Etcher `1.7.1 (1.7.1)`.

```shell
MacBook-Pro ➜ pwd
/Applications/balenaEtcher.app/Contents/MacOS

MacBook-Pro ➜ file balenaEtcher
balenaEtcher: Mach-O 64-bit executable x86_64
```

fishcharlie, 2021-12-06:
@thundron Any chance you could please provide a status update here? 😃

thundron, 2021-12-08:
@fishcharlie sure! unfortunately there's not much information yet that we can safely point to as the root cause; there have been *some* changes to our CI system, but none of those changes should've impacted the universal build in any shape or form; moreover, there have been some dependencies updates but again, none of those dependencies/changes were pertaining to the macOS universal package, or the package build at all (if not for how dependencies are packed), so we're still investigating!

fishcharlie, 2021-12-09:
@thundron Thanks for the update.

I decided to look into this a little bit, and compared the versions between https://github.com/balena-io/etcher/compare/v1.5.121...v1.5.122.

The only thing that stands out is the resin submodule update: https://github.com/balena-io/scripts/compare/d1b05ad312e65ea82b1c16b31f5af3c0b5fa2777...8dfa21cfc23b1dbc0eaa22b5dbdf1f5c796b0c2c. Everything else looks completely unrelated to this issue.

So I dug into those changes. The thing that stands out to me is if you search for `platforms` on that page, I see a lot of changes related to that within the docker/build.sh file. Is that related? Maybe the platforms are excluding Apple Silicon, where before it was undefined so it would build for all platforms? No idea. All questions that I think are above my head at this point, and without digging into why those changes were made, and what some of those variables are set to, it's difficult to know why things are failing.

Makes me wonder if reverting that submodule update and attempting again, would produce an Apple Silicon build. That would at least narrow things down. But that raises other questions that I'm not sure about, such as: were there behind the scene changes to the CI system where these submodule changes are required now?

I might try to dig into this a bit deeper later on and try to run some of the things myself. But I'm pretty unfamiliar with the CI system here. Loading https://ci.product-os.io/builds/95676, produces a spinning Loading page that never completes. But that doesn't even look like a recent deployment build necessarily.

However, it sounds like @thundron probably has a **lot** more context here about what is going on than I do. So I'm not sure me spinning my wheels on it will produce a solid outcome and solve the issue.

Wish I could help more. Really want to support the open source community and get this resolved for everyone (and myself 😉). But @thundron if there is anything else I can do to support this effort, please let me know. Would be happy to try to run some of this locally and debug further, but might need some support in getting pointed in the right direction to get that setup.

thedocbwarren, 2021-12-12:
Simple solution could be to publish two builds for now (also on homebrew.)

n0rt0nthec4t, 2021-12-31:
1.73 and STILL no universal binary..

robd003, 2022-01-07:
> @fishcharlie sure! unfortunately there's not much information yet that we can safely point to as the root cause; there have been _some_ changes to our CI system, but none of those changes should've [...]

@thundron Could you share the CI build configuration? I think there's enough motivated people here to get it working again if we can get the details.

danielktdoranie, 2022-01-19:
Okay, so still no Native support of Apple Silicon? We had working Apple Silicon support in 1.5.121... what happened? Why was this dropped?

fishcharlie, 2022-01-19:
@danielktdoranie Wasn't dropped intentionally. See the previous comments. @thundron mentioned that although some CI changes were made, nothing should have impacted Apple Silicon support. Although clearly something did.

I posted a summary of some work I did to pinpoint the issue above.

All of your questions have been previously answered in this thread tho.

danielktdoranie, 2022-01-19:
> @danielktdoranie Wasn't dropped intentionally. See the previous comments. @thundron mentioned that although some CI changes were made, nothing should have impacted Apple Silicon support. Although [...]

And yet we are still without native Apple Silicon support.  Last post was over a month ago.

Intel Macs aren’t being made any longer so any Intel code for Macs at this point is by definition legacy support. 

Instead of trying to figure out why the universal binary doesn’t work, why not just release an arm64 Mac version on the Releases page?

fishcharlie, 2022-01-19:
@danielktdoranie

> Last post was over a month ago.

Kinda hard to say this when a LOT of people take a lot of time off during December...

> Intel Macs aren’t being made any longer so any Intel code for Macs at this point is by definition legacy support.

This is categorically false. Just because Apple hasn't released any new Intel Macs, doesn't mean they aren't being made any longer. Intel Macs are still being manufactured to this day. And more importantly remain for sale from Apple.

> Instead of trying to figure out why the universal binary doesn’t work, why not just release an arm64 Mac version on the Releases page?

Making an educated guess here. The amount of effort that would require would likely be just as substantial as fixing the issue.

---

Remember. This is open source software. Normally I'd be very sympathetic to your case. Pushing companies to stay current with technology is critical.

However, it is open source. You can go and create your own build for native support just fine. No issues. In fact I've done that myself.

Or you can spend the time required working to fix or pinpoint the issue yourself.

It's not like this project has seen a lot of major updates recently and this is just on the back burner. It clearly looks like a resources issue, not a priority issue. Which makes complaining about it unproductive.

danielktdoranie, 2022-01-19:
The Mac Pros are the only Intel Macs left and they’ll be replaced with an M1 counterpart soon enough, even Apple said themselves that they will transition to Apple Silicon completely with in 2 years, and we are only a few months out from that. Also that Mac Pro design was released in December 2019: so yeah as NO NEW Intel macs are being released Intel Mac code is is de facto legacy support.

My comments are no more “unproductive” than your’s: Take your own advice.

nodesocket, 2022-01-19:
Ok, gentlemen let’s stop the petty bickering.  However, Intel Macs are basically end of life and Apple did say it would be a two year transition to Apple Silicon which looks more like three years now. Apple Silicon support is a must have, since I suspect Apple will be sunsetting Rosetta 2 shortly.

Let’s focus on getting the universal binary build working instead of adding complexity for a dedicated arm64 build.

It should be fairly straightforward to git blame and go line by line. Also, look at the 3rd party build tooling to see if they changed any API’s or anything. My concern is perhaps this is not a regression, but intentional. Seems suspicious a regression would remove universal binary support and go unfixed for months. Perhaps running universal binary blocks something the are currently doing (telemetry) or worse?

danielktdoranie, 2022-01-19:
> Let’s focus on getting the universal binary build working **again** instead of adding complexity for a dedicated arm64 build.

I understand that ultimately that is the best solution for legacy Intel Mac users as well as Apple Silicon users, my thinking was if a stop-gap Apple Silicon native only build could be complied and put on the releases page it would help Apple Silicon users like myself… given we are months in at this stage.

I’d be happy for either a native or universal binary:

thedocbwarren, 2022-01-28:
> > Let’s focus on getting the universal binary build working **again** instead of adding complexity for a dedicated arm64 build.
> 
> I understand that ultimately that is the best solution for legacy Intel Mac users as well as Apple Silicon users, my thinking was if a stop-gap Apple Silicon native only build could be complied and put on the releases page it would help Apple Silicon users like myself… given we are months in at this stage.
> 
> I’d be happy for either a native or universal binary, just don’t want to run Rosetta 2. It’s a memory and battery hog. I have removed it from all of my Apple Silicon Macs.

I'd be happy to be able to compile myself if the procedure were available.

n0rt0nthec4t, 2022-02-22:
so 1.7.6 released.. STILL NO UNIVERSAL binary back.

danielktdoranie, 2022-02-22:
> so 1.7.6 released.. STILL NO UNIVERSAL binary back.

Clearly a huge priority for the team

thedocbwarren, 2022-02-22:
Thinking out loud, this sounds like time to look at alternatives.  This feature can be accomplished with 'dd.'

Also, I wish this could work as well as Ubuntu's Startup Disk Creator that allows a test with QEMU

n0rt0nthec4t, 2022-02-23:
> > so 1.7.6 released.. STILL NO UNIVERSAL binary back.
> 
> Clearly a huge priority for the team

Clearly they don't give a F*( to fix it and they CAUSED the issue in the first place

dtischler, 2022-02-24:
Thank you everyone for your patience while we research, understand, and plan a path forward for this open-source project.  As pointed out by some commenters, this is a free product, and with the source code provided in this repo, you can certainly build, fix, improve, or iterate a version locally that meets your needs.

As [electron builder](https://www.electronjs.org/blog/apple-silicon#what-do-i-need-to-do) provides a simple way to build for multiple platforms from one place, [native modules](https://www.electron.build/multi-platform-build) can be built only on the same platform as the target.  Etcher uses multiple native modules.

To have a working Mac M1 (Arm64) we will either need to replace those native modules, or add an M1 Mac to our build pipeline.  The universal build is just packaging the x64 and arm64 together, also doubling the size.

If you are looking to perform a native Arm64 build on your M1, the process would look like: 

 - Install Xcode
 - Git clone this repo
 - `npm i`
 - Check if all native modules “node-gyp”-ed correctly ... If not go to the folder of the module (inside `node_modules`) and run `electron-rebuild`
 - `npm run webpack`
 - `./node_modules/.bin/electron-builder`
 - (no additional parameters are needed, as you are running on the target architecture)
 - You output .dmg file will be located in the `dist` folder

thedocbwarren, 2022-02-25:
Thank you!  I'm going to try this out and build it and report back.  You mention native modules.  Which ones do we need to work with?  Is this node-gyp, node-sass, etc. or are there major other modules required that need a build from source on M1?  My intention to to build 100% non-rosetta native app.

thedocbwarren, 2022-02-25:
Also, are you open to release contributions for this platform?  I'm a founder / maintainer of an opensource project(s) and can build these for you and contribute back.

thedocbwarren, 2022-02-25:
ok works perfectly.  I will need to enable signing and some other stuff for real releases but it works without any issue on my MacBook Pro 13 M1.

dtischler, 2022-02-28:
@thedocbwarren -- Excellent work!  We are determining the best way forward here, and will report back to you shortly!  :-)

mcraa, 2022-02-28:
Thank you @dtischler, @thedocbwarren @fishcharlie, and everyone.
Created and pinned issue #3714 , we will update the list with your build @thedocbwarren if you are willing to share.

thedocbwarren, 2022-02-28:
> Thank you @dtischler, @thedocbwarren @fishcharlie, and everyone. Created and pinned issue #3714 , we will update the list with your build @thedocbwarren if you are willing to share.

You bet!  how would you wish for me to submit it?

thedocbwarren, 2022-02-28:
oh and how do you want to handle the signing? I think it's less of an issue since it's not on the Mac App store.

lurch, 2022-03-01:
> Created and pinned issue https://github.com/balena-io/etcher/issues/3714

@mcraa I guess the third-party Solus, Arch/Manjaro, Brew and Chocolatey builds mentioned in https://github.com/balena-io/etcher/blob/master/README.md ought to be added to that issue? :man_shrugging: 

Also, #3714 says "Simply send us a message or **comment** with the link and description" but people can't add a comment to that issue as it's locked :wink:

robd003, 2022-03-01:
It looks like this pull request might have M1 builds working https://github.com/balena-io/etcher/pull/3712

mcraa, 2022-03-01:
@robd003 that PR is just an explicit config for electron builder to make arm64 output for mac.
Same result as running it with the `-m` option or on an M1 mac without parameters.

@lurch good point I was thinking about keeping that issue clean from clutter and submissions can come in other issues or different channels. I will probably reopen it.

@thedocbwarren 
 - signing
    if you have a key you may sign it or leave it unsigned, there is the note in #3714 that the builds listed there are not made by balena, anyone who downloads should be aware
- how to submit
  as lurch pointed out I will unlock the issue, you can comment your link there or here, as suggested in #3714 you may create a repo on github for it, and add releases there, or if you have any other hosting preference, it is your choice

thedocbwarren, 2022-03-02:
Link to my community build (also shared in other issue.). https://github.com/Augmentedjs/balena-io-etcher-builds/releases/tag/v1.7.7

thedocbwarren, 2022-03-02:
I can also build (and will attempt to do so) Arm64 on Ubuntu.  I may even try Windows 11 Arm.

lurch, 2022-03-03:
@thedocbwarren It's probably worth editing your post in the other issue to provide a bit more info e.g. which OS / CPU architecture / etc. it's built for?

robd003, 2022-03-15:
@thedocbwarren I'm getting a weird error when trying to launch your build.

```
"balenaEtcher" is damaged and can't be opened. You should move it to the Trash.
```

If I try to run it from the console this is the error output: The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-128 "userCanceledErr" UserInfo={_LSLine=3845, _LSFunction=_LSOpenStuffCallLocal}

thedocbwarren, 2022-03-15:
> @thedocbwarren I'm getting a weird error when trying to launch your build.
> 
> ```
> "balenaEtcher" is damaged and can't be opened. You should move it to the Trash.
> ```
> 
> If I try to run it from the console this is the error output: The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-128 "userCanceledErr" UserInfo={_LSLine=3845, _LSFunction=_LSOpenStuffCallLocal}
> 
> This is on macOS 12.3 arm64

Let me check on that. It might be a signing error (which I'm going to correct soon.)

thedocbwarren, 2022-03-19:
New build for Apple Silicon v1.7.8 -> https://github.com/Augmentedjs/balena-io-etcher-builds/releases/tag/v1.7.8

robd003, 2022-03-19:
@thedocbwarren Awesome! Downloading it now, I'll let you know how it goes.

robd003, 2022-03-19:
@thedocbwarren Unfortunately I'm getting the same error.

However if you manually run `xattr -rc /Applications/balenaEtcher.app` then it'll work.

thedocbwarren, 2022-03-19:
Ok, I'll work on this.  I think it's signing.  I promise I'll fix this.

thedocbwarren, 2022-03-27:
Signing complete. Release 1.7.8 is now signed.

mcraa, 2022-04-23:
Strange @thedocbwarren, there were no changes effecting the mac build. 
Only the fix for .deb install.
At the same time it means not having 1.7.9 on mac won't make any mac user miss out on anything, it should be identical to 1.7.8

thedocbwarren, 2022-04-23:
> Strange @thedocbwarren, there were no changes effecting the mac build. Only the fix for .deb install. At the same time it means not having 1.7.9 on mac won't make any mac user miss out on anything, it should be identical to 1.7.8

I saw that.  I'll keep on it and see what could be causing it.  I'll check with Debian as well to see what could be the cause.  I reviewed the webpack config and didn't see anything that prevents the dist folder from getting created or updated for output.

thedocbwarren, 2022-04-23:
> Strange @thedocbwarren, there were no changes effecting the mac build. Only the fix for .deb install. At the same time it means not having 1.7.9 on mac won't make any mac user miss out on anything, it should be identical to 1.7.8

Ok, confirmed Debian does not output anything as well.

thedocbwarren, 2022-04-29:
> Strange @thedocbwarren, there were no changes effecting the mac build. Only the fix for .deb install. At the same time it means not having 1.7.9 on mac won't make any mac user miss out on anything, it should be identical to 1.7.8

I am unable to find where the issue is.  Something may be different in the output process -or- something is going wrong in webpack's config to not produce a 'dist' folder.  I'm less concerned about 1.7.9, as stated, doesn't do much for non-deb platforms but more for 1.7.10 or later.

thedocbwarren, 2022-11-07:
1.7.12 builds fine for MacOS ARM, however like before the build script does not generate a DMG.  Could I get new instructions for how to build this correctly?

nodesocket, 2022-11-07:
Just installed the latest version `1.7.15` on macOS Ventura, but same thing. Running as `intel` emulated via Rosetta.

[image]

thedocbwarren, 2022-11-08:
Yes, @nodesocket it needs a native build.  I'll wait for the release storm to finish and I'll try again to build.

thedocbwarren, 2022-11-16:
1.10.0 builds, but produces no package (DMG.) Not sure how to produce to publish a community build for Apple Silicon.

lurch, 2022-11-16:
Looks like https://github.com/balena-io/etcher/blob/master/.github/actions/publish/action.yml#L173 _might_ be where the magic happens?

thedocbwarren, 2022-11-18:
@lurch What I can see with this (maybe wrong) is it's using GitHub actions to build that, meaning local builds won't produce a DMG.  This seemed to be true from 1.7.9 on.  I can launch via CLI to run the electron app but not install it.  If this is true,  I can't build a DMG to publish and they will have to build this officially in their CI.

thedocbwarren, 2022-11-18:
I want to add, I saw somewhere those actions are x64 only so that is an issue for ARM.  But just to mention, this is open-source and not being able to build and install kind of breaks that and should be corrected.  Unless the new procedure is to create a start script and build and run from npm I suppose.

thedocbwarren, 2023-03-15:
I have been able to compile recently but no DMG is built.  Is GitHub actions the only way to build a DMG with the codebase?

donluca, 2023-05-22:
> * Install Xcode
> * Git clone this repo
> * `npm i`
> * Check if all native modules “node-gyp”-ed correctly ... If not go to the folder of the module (inside `node_modules`) and run `electron-rebuild`
> * [...]

I'm trying to build an arm64 native build, but I'm encountering this error after doing `./node_modules/.bin/electron-builder` :

```  ⨯ The appleIdPassword property is required when using notarization with appleId  failedTask=build stackTrace=Error: The appleIdPassword property is required when using notarization with appleId```

I don't care about distributing it, I just want it for myself, how can I tell electron to skip notarization?

thedocbwarren, 2023-07-12:
Getting the same error but I can sign.  What is the procedure?  I can start building the community Apple Silicon builds again.

dfunckt, 2023-07-13:
Closing in favor of #3728

[6 more replies]

closed 2023-07-13 by dfunckt