balena-io/etcher #3595: Etcher DEB and AppImage will not launch see details!

ANKOU66, 2021-09-07:
On latest development UBUNTU Impish BUILD 21.10 , Etcher DEB and AppImage will not launch.

thundron, 2021-09-08:
@ANKOU66 what version of etcher did you try? can you try another one and report that too?

cstrahan, 2021-09-19:
Same here. Had an older version installed (forget which), which did work. Latest does not work -- no windows display. However, the process are hanging around:

```
cstrahan 3928133  0.0  0.0   9484  3156 pts/17   S+   00:11   0:00 /bin/bash /usr/bin/balena-etcher-electron
cstrahan 3928136  1.5  0.1 4689940 137036 pts/17 Sl+  00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin
cstrahan 3928139  0.0  0.0 211040 52932 pts/17   S+   00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin --type=zygote --no-zygote-sandbox
cstrahan 3928140  0.0  0.0 211040 53008 pts/17   S+   00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin --type=zygote
cstrahan 3928142  0.0  0.0 211040  9952 pts/17   S+   00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin --type=zygote
cstrahan 3928173  1.3  0.0 336164 90072 pts/17   Sl+  00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin --type=gpu-process --field-trial-handle=1021344705118950845,8250590479023267773,131072 --enable-features=WebComponentsV0Enabled --disable-features=CertVerifierService,CookiesWithoutSameSiteMustBeSecure,SameSiteByDefaultCookies,SpareRendererForSitePerProcess --gpu-preferences=UAAAAAAAAAAgAAAQAAAAAAAAAAAAAAAAAABgAAAAAAAwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgAAAAAAAAACAAAAAAAAAAAAAAAAAAAAAIAAAAAAAAAAgAAAAAAAAACAAAAAAAAAA= --shared-files
cstrahan 3928177  0.0  0.0 277728 66220 pts/17   Sl+  00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin --type=utility --utility-sub-type=network.mojom.NetworkService --field-trial-handle=1021344705118950845,8250590479023267773,131072 --enable-features=WebComponentsV0Enabled --disable-features=CertVerifierService,CookiesWithoutSameSiteMustBeSecure,SameSiteByDefaultCookies,SpareRendererForSitePerProcess --lang=en-US --service-sandbox-type=network --shared-files=v8_context_snapshot_data:100
cstrahan 3928250  0.0  0.0 284648 29456 pts/17   S+   00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin --type=broker
cstrahan 3928289  1.6  0.0 48604420 121908 pts/17 Sl+ 00:11   0:00 /opt/balenaEtcher/balena-etcher-electron.bin --type=renderer --field-trial-handle=1021344705118950845,8250590479023267773,131072 --enable-features=WebComponentsV0Enabled --disable-features=CertVerifierService,CookiesWithoutSameSiteMustBeSecure,SameSiteByDefaultCookies,SpareRendererForSitePerProcess --lang=en-US --app-path=/opt/balenaEtcher/resources/app --no-sandbox --no-zygote --num-raster-threads=4 --enable-main-frame-before-activation --renderer-client-id=5 --no-v8-untrusted-code-mitigations --shared-files=v8_context_snapshot_data:100
```

No stdout to share :(