balena-io/etcher #3601: Is this common knowledge that `rpiboot` is needed in order to flash images to BalenaFin?

doronbehar, 2021-09-21:
- **Etcher version:** 1.5.121
- **Operating system and architecture:** NixOS
- **Image flashed:** balena-cloud-fincm3-2.80.3+rev1-dev-v12.7.0.img.zip
- **Do you see any meaningful error information in the DevTools?** Irrelevant

So I bought a BalenaFin a few months ago, and only now I got the time to use it. I asked a question on the forums but I got no reply:

https://forums.balena.io/t/balena-etcher-wont-detect-my-balena-fin-on-a-different-computer-it-does/348017

Somehow I've found out that I need to run `rpiboot` in order for my Balena Etcher to detect the compute module. This issue is a complaint about documentation - is this documented somewhere? Could it be that Etcher should run `rpiboot` on it's own and it doesn't do it for me?

lurch, 2021-09-21:
Maybe this helps? https://github.com/balena-io/etcher/issues/3205#issuecomment-902434754

doronbehar, 2021-09-21:
> Maybe this helps? [#3205 (comment)](https://github.com/balena-io/etcher/issues/3205#issuecomment-902434754)

As explained in [the thread at the forums](https://forums.balena.io/t/balena-etcher-wont-detect-my-balena-fin-on-a-different-computer-it-does/348017), I did try launching Etcher with and without sudo and it didn't detect the device unless I ran `rpiboot`.