balena-io/etcher #3606: Oops! Looks like the flash failed. Somethin went wrong while writing ..

tolginho, 2021-10-01:
ETCHER v1.5.122, v.1.5.121, v.1.6.0
Windows 10, x64 i7-6820HQ 2.7GHz
all different kind of images

{"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)\n    at TLSSocket.emit (events.js:315:20)\n    at TLSSocket._finishInit (_tls_wrap.js:932:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)","config":{"url":"https://balena.io/etcher/static/config.json","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.6.0"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"json","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1,"transitional":{"silentJSONParsing":true,"forcedJSONParsing":true,"clarifyTimeoutError":false}},"code":"CERT_HAS_EXPIRED"}
console.<computed> @ gui.js:51281
gui.js:51281 {"stack":"TypeError: Cannot read property 'map' of undefined\n    at file:///C%3A/Users/TOHISIR/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:73469:30\n    at Animator.update (file:///C%3A/Users/TOHISIR/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:51180:35)\n    at Animator.updateOrSkip (file:///C%3A/Users/TOHISIR/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:51172:44)\n    at file:///C%3A/Users/TOHISIR/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:51160:30","message":"Cannot read property 'map' of undefined"}
console.<computed> @ gui.js:51281

[image]

lurch, 2021-10-01:
> "message":"certificate has expired","stack":"Error: certificate has expired\n at TLSSocket.onConnectSecure

Possibly related to https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/ ? Although it seems a shame that Etcher isn't able to "ignore" that error and continue to flash the image regardless?

tolginho, 2021-10-01:
[image]
maybe this will help u

lurch, 2021-10-01:
> "Error: EPERM: operation not permitted, write"

Have you accidentally write-protected your SD card? Have you tried running Etcher as Administrator?

thundron, 2021-10-01:
absolutely did @tolginho , the error is about missing permissions. can you check whether you have Controlled Folder Access active? you also need an admin account (or admin permissions) to flash, so maybe running as admin could also help (provided you have the above check disabled/whitelisted)

tolginho, 2021-10-04:
Hello, i already tried to run balena as admin, still doesnt work. Controlled folder access is active. Unfortunately, I cannot add balena as a valid app because I have a company laptop that refuses to do so. My problem is that everything was working fine a week ago (even without running balena as admin).

I just wanted to give you console logs when running balena as admin but the developer window doesnt show up with the given shortcut (cmd shift i /cmd alt i) when running as admin, is this problem known?

lurch, 2021-10-04:
> My problem is that everything was working fine a week ago (even without running balena as admin).

Is it possible that there's been some behind-the-scenes policy update on your corporate network, that has (silently) locked down the restrictions even tighter? :shrug:  :lock:

cbabos, 2021-10-16:
Not sure whether this can help as this is for Windows, on OSX (Big Sur) I had the similar if not the same issue. 
There after hours of trying with Folder access I've tried Full Disk Access which was finally resulting in a successful flashing (to SSD)