balena-io/etcher #3607: Problems Using Windows ISO when booting a Pendrive.

vitor2424, 2021-10-01:
- **Etcher version 1.5.109** 
- **Linux Mint 20.2 Uma ** 
- **Win10_21H1_BrazilianPortuguese_x64.iso** 

  I am using Linux or a while now, in general I like it, but when playing games that aren't on Steam, It Sucks!
So, I was trying to do a dual boot Linux/Windows10, and when I tried to use the iso file that microsoft gives on their oficial site, This message pops on the screen: "Missing partition table. It looks like this is not a bootable image. The image does not appear to contain a partition table, and might not be recognized or bootable by your device." 
  Can anyone give a light in this issue? Obs: I am runing Etcher By a AppImage, I don't if it matters, but if it do, then there it is.

lurch, 2021-10-02:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

thundron, 2021-10-02:
You'll have to use a different tool for flashing Windows images and/or perform extra operations that you'll likely find somewhere online about how to set-up a newly-flashed Windows USB to also be a bootable installer.

closed 2021-10-02 by thundron