balena-io/etcher #3624: creating image from sd or cloning enhancement request

blacknell, 2021-11-01:
I'm using the latest 1.5.122 and when cloning, there is no way to select the same SD drive as the target. I only have one SD card slot on my Mac.

So either it could be enhanced to allow this, or could it be enhanced to save the resulting image (relates #3232) which I need anyway since I will be cloning this drive to others in the unforeseen future.

lurch, 2021-11-01:
Perhaps the easiest "workaround" for now is to just buy a USB SD card-reader? :wink:

alex-ssom, 2021-11-12:
Just increase the block size to something like 4 Mb `bs=4M`. You can also use the `status=progress` argument to get some feedback on the progress of the copy operation.

closed 2022-08-27 by blacknell

ReanimationXP, 2026-09-22:
This request was closed as completed without being completed also, as I cannot select the same source drive as a destination, which would provoke the saving of an image file, even if it's to a location I can't designate.