balena-io/etcher #3626: Flash from URL fails with "Error: certificate has expired"

steev, 2021-11-02:
- **Etcher version:** 1.6.0
- **Operating system and architecture:** Windows 10 x86_64
- **Image flashed:** kali-linux-2021-W43-rpi0w-armel.img.xz
- **Do you see any meaningful error information in the DevTools?** 

gui.js:51281 {"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)\n    at TLSSocket.emit (events.js:315:20)\n    at TLSSocket._finishInit (_tls_wrap.js:932:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)","config":{"url":"https://balena.io/etcher/static/config.json","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.6.0"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"json","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1,"transitional":{"silentJSONParsing":true,"forcedJSONParsing":true,"clarifyTimeoutError":false}},"code":"CERT_HAS_EXPIRED"}

{"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)\n    at TLSSocket.emit (events.js:315:20)\n    at TLSSocket._finishInit (_tls_wrap.js:932:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)","config":{"url":"https://arm.kali.org/kali-weekly/kali-linux-2021-W43-rpi0w-armel.img.xz","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.6.0"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"stream","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1,"transitional":{"silentJSONParsing":true,"forcedJSONParsing":true,"clarifyTimeoutError":false}},"code":"CERT_HAS_EXPIRED"}

I believe this has to do with the LetsEncrypt root certificate expiration as the server I am using uses a LetsEncrypt cert, however the site works fine in browsers, and I can download the file over https just fine.

thundron, 2021-11-02:
We're working towards resolving this issue but it will require a bit more time unfortunately!

Tortoise17, 2021-11-17:
I am also facing it. Now no version of etcher is working apparently.

richardwellerson, 2021-11-29:
Same problem at version 1.7.1 on Ubuntu 20.04 x64.

richardwellerson, 2021-11-29:
[image]

Well, my devTools shows a error about "message"... But apparently my flash disk works perfectly...

orayb, 2022-01-01:
Same problem with 1.7.3-x64, running on Cent OS 7

cascadeorca, 2022-01-14:
Issue is happening with some IMGs, but not all.

`gui.js:46 {"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)\n    at TLSSocket.emit (events.js:315:20)\n    at TLSSocket._finishInit (_tls_wrap.js:932:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)","config":{"url":"https://balena.io/etcher/static/config.json","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.7.3"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"json","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1,"transitional":{"silentJSONParsing":true,"forcedJSONParsing":true,"clarifyTimeoutError":false}},"code":"CERT_HAS_EXPIRED"}`

Danahi, 2022-01-25:
I am faced with the same issue on windows 10.
One time I could flash an image. After that it failed.
```
{"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)\n    at TLSSocket.emit (events.js:315:20)\n    at TLSSocket._finishInit (_tls_wrap.js:932:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)","config":{"url":"https://balena.io/etcher/static/config.json","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.7.3"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"json","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1,"transitional":{"silentJSONParsing":true,"forcedJSONParsing":true,"clarifyTimeoutError":false}},"code":"CERT_HAS_EXPIRED"}
console.<computed> @ gui.js:46
```

Tank1334, 2022-01-27:
Also having this problem

FirstWhiteHacker, 2022-02-11:
If you change https to http, it will succeed.

steev, 2022-02-11:
> If you change https to http, it will succeed.

Well yes, because then it's not trying to verify the certificate, because it's not a *s*ecure connection.  Not all web servers expose http these days however.

RodBarnes, 2022-02-14:
I get these same errors but it doesn't appear to have any affect on the operation.  The flashed drives I've created seem to work fine.

closed 2023-07-12 by dfunckt