balena-io/etcher #3627: "Priviledge access in order to flash disks" prompt confusing to user when smartcard PIN is used for auth

tecandrew, 2021-11-04:
- **Etcher version:**  1.6.0
- **Operating system and architecture:** macOS Big Sur amd64
- **Image flashed:** Pop!_OS Intel
- **Do you see any meaningful error information in the DevTools?** nope

When a smartcard is used to login, or authenticate, on macOS for flashing disks it asks for the admin password. Every attempt failed until I realized the smartcard (yubikey in this case) was blinking green. I entered the PIN, instead of the admin password, and it worked fine. 

### Error Message

```
balena etcher needs priv... access in order to flash disks. Enter password
```

### Recommendation

It should really say what method of auth is being used like macOS's "enter PIN" or "enter password" for login at the lock screen.

| pswd | pin | 
|---|---|
|[image] |[image] |

Nmatt1, 2022-01-23:
This is a simple issue really. 

With Etcher you need admin permissions to make changes to drives, such as writing to and formatting. In MacOS when you are formatting a drive in disk utilities it asks for password before you make any changes, this is just to confirm you are an admin. Its the same here.

tecandrew, 2022-01-25:
it's been a few months since I filed this, but I filed this as a recommendation / feature-request. I was fiddling with Etcher for about 30 minutes before I realized Etcher really meant to ask me for my PIN, not my password.

user error I guess? I usually leave my yubikey inserted since I use it for many auths.

joosters, 2022-07-28:
I ran into this same issue. I don't think it is user error, the dialog box specifically asks for a *password*. The standard system system auth dialog (used by other programs) changes the text when you are using a yubikey to make it clear that you need to type in your PIN and not your password. It's a bug that Etcher's custom dialog box doesn't state that you need to type your PIN.

tecandrew, 2022-07-28:
> It's a bug that Etcher's custom dialog box doesn't state that you need to type your PIN.

Yeah, I figured this was the issue. I don't know enough of the code base to do this myself and submit a PR, but it seems straightforward. Hopefully someone can pick this up to implement the resolution

lurch, 2022-07-28:
Possibly related to https://github.com/balena-io-modules/sudo-prompt ? :shrug:

tecandrew, 2022-07-28:
Had no idea etcher used it's own tool and it's in _that_ repo. Hmm, this issue should have probably been listed there instead :sweat_smile:

dfunckt, 2023-07-12:
Closing in favor of #3321

closed 2023-07-12 by dfunckt