balena-io/etcher #3635: BE v1.70 Portable Does Not run when executed

maxxxis, 2021-11-18:
- **Etcher version:** - 1.07p
- **Operating system and architecture:** Win10x64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** 
No, when I execute the file, it shows as consuming cpu resources, but it never displays an interface. I've resorted to using the install version, but would prefer the portable one. Unsure what other diagnostics I can provide for this situation.

duckimann, 2021-11-18:
I though i have the same issue but what actually run, just too long after the user click executable file. like 5 minutes after i click the executable.

lurch, 2021-11-18:
Perhaps an overly-aggresive virus-scanner or something? :shrug:

duckimann, 2021-11-18:
> Perhaps an overly-aggresive virus-scanner or something? 🤷

I only have the Windows Defender lol

cybermcm, 2021-11-22:
Same here, Windows version ansible_distribution_major_version, BE ver 1.7.0
If Defender is active, program start is very slow, BE and Defender using massive CPU time.
If Defender is disabled → BR starts faster but still waiting time from start to first window popup (running on a 5-year-old PC)

KaMyKaSii, 2021-11-24:
> I though i have the same issue but what actually run, just too long after the user click executable file. like 5 minutes after i click the executable.

Same here. Windows 11 x64 with all updates installed. I don't use any third party antivirus program and Windows Defender is disabled thorugh Defender Control

redking974, 2021-11-26:
Same here. W10 updated. I see the ressources on task manager but no gui