balena-io/etcher #3638: A Blank white screen is visible because of 'Module did not self-register' Error in console window

prashantbodh099, 2021-11-21:
- **Etcher version: v1.7.0** 
- **Operating system and architecture: Windows 10 , AMD 64** 
- **Image flashed: None** 
- **Do you see any meaningful error information in the DevTools?: Module did not self-register: '\\?\D:\etcher-master\generated\modules\etcher-sdk\node_modules\drivelist\build\Release\drivelist.node'.** 

I wanted to build the etcher.exe after making some changes in the source code of balena-etcher app. So I cloned the repo into my windows system and followed the steps mentioned in [contribution-doc](https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md).
Somehow, 

> make electron-build

command was not able to build all the node modules. So I build some node modules (including drivelist) manually and It worked well. 

> npm run webpack

worked fine without any errors. But when I launched the etcher application from the the developer command prompt of VS 2019, I could only see a blank white screen with an error in console. 

[image]
[image]

Note : I haven't done any changes in the source code of etcher yet. I got this error during development setup only.

lurch, 2021-11-22:
It's been **many** years since I've tried building Etcher myself, but it sounds like you might have built drivelist against the Node ABI version instead of the Electron ABI version? See many similar problem-reports at https://github.com/balena-io-modules/drivelist/issues
I think rather than "build some node modules (including drivelist) manually" you need to try and get the "make electron-build" properly working? :man_shrugging:

prashantbodh099, 2021-11-23:
Yeah. Actually, "make electron-build"  was giving errors like "missing drivelist.node", So I build them manually. Now, I tried to set up the development environment on some other system and it is working fine there.