balena-io/etcher #3639: GPU process isn't usable. Goodbye.

zmiimz, 2021-11-23:
- **Etcher version:** 1.7.0
- **Operating system and architecture:**  openSUSE Tumbleweed 20211005, Linux-x86_64
- **Image flashed:**  no, cannot start Etcher
- **Do you see any meaningful error information in the DevTools?**  not available

./balenaEtcher-1.7.0-x64.AppImage 
libva error: /usr/lib64/dri/i965_drv_video.so init failed
libva error: /usr/lib64/dri/i965_drv_video.so init failed
libva error: /usr/lib64/dri/i965_drv_video.so init failed
[Axios v1.7.0] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.0] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.0] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[19146:1123/180729.228900:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenanM6ZuF/balena-etcher-electron: line 10: 19146 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"

PC has Nvidia video card and NVIDIA Driver Version 470.74 installed.

NikitinGs, 2021-11-28:
It is one of tumbleweed's problem with nvidia. To run most of electron apps you should add flag --disable-gpu-sandbox

thundron, 2021-11-29:
As @NikitinGs said. Closing

closed 2021-11-29 by thundron

zmiimz, 2021-11-29:
I have no problem with Nvidia on Tumbleweed, environment and driver are used intensively every day with CUDA, and many intensive 3D applications etc. I believe it is not an issue experienced exclusively with openSUSE, this seems to be an issue with electron. But anyway, thanks for the recipe.

ERROR001001, 2022-05-09:
> 

how do I add flags and wdym add a flag? (Sorry, I'm new to linux)

NikitinGs, 2022-05-10:
> > 
> 
> how do I add flags and wdym add a flag? (Sorry, I'm new to linux)

launch it from console and add flags after space (example: `./balenaEtcher-1.7.9-x64.AppImage --disable-gpu-sandbox` if you use appimage version). Or, if you are use shortcuts, add them to launch options, like you do on windows

headkaze, 2022-06-12:
Create a file called etcher.desktop with the contents below:
```ini
[Desktop Entry]
Name=Etcher
Comment=Flash OS images to SD cards & USB drives, safely and easily
Exec=sh -c "/home/username/Applications/balenaEtcher.AppImage --disable-gpu-sandbox"
Icon=/home/username/Applications/Icons/etcher.png
Type=Application
StartupNotify=false
Terminal=false
Categories=DiskTools;Utility
```

Copy etcher.desktop to /usr/share/applications

You can get the icon from [here](https://upload.wikimedia.org/wikipedia/commons/2/2d/Etcher-icon.png)

linus-ha, 2022-08-04:
The `--disable-gpu-sandbox` param should be included in the `.rpm`'s  `.desktop` file by default, as Balena Etcher is not working on RHEL and derivatives (e.g, Fedora) when the NVIDIA driver is used. Wayland is used as the default, so this is a major problem.

In addition, I don't think that GPU acceleration is really needed for etcher in general. Of course this should be removed again, when NVIDIA got everything working.

HughPH, 2022-08-18:
Failing on Ubuntu 22.04 too. I agree, if it's not working on NVidia generally, that's 50% of users. Balena should deploy with the flag until this is resolved.

david2628, 2022-09-17:
This is still a problem on Arch (installed with yay) 
i Agree to please put in the Flag

NicTanghe, 2022-10-31:
Its verry stupid no to have  --disable-gpu-sandbox as the default and enable-gpu-sandbox as a flagg

linus-ha, 2022-10-31:
@NicTanghe I do not think this is stupid in general, as using the GPU could increase performance for other applications. In this case though, "--disable-gpu-sandbox" should obviously be the default, as not having it set by default causes problems and is not really necessary for this kind of application.

NicTanghe, 2022-10-31:
right but isn`t sandboxing the gpu something that is a part of this application ?

yuwanlin, 2023-02-28:
Are there any security risks when --disable-gpu-sandbox ?

aprilandjan, 2023-10-17:
> --disable-gpu-sandbox

I'm wondering why is the situation. If that is the truth, then electron/chromium team will probably make it as the default best practice 🤔