balena-io/etcher #3640: Missing directory in /opt during install - solved by symlink.

dmcardlenl, 2021-11-24:
- **Etcher version:** 1.5.117
- **Operating system and architecture:** Fedora 34 x86_64 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Missing directory in /opt during install - solved by symlink.
/usr/bin/balena-etcher-electron points to /opt/balena-etcher-electron/balena-etcher-electron
in /opt ln -s ./balenaEtcher ./balena-etcher-electron needed - or similar - YMMV.

lurch, 2021-11-24:
See also #3486

dmcardlenl, 2021-11-24:
Yes, I didn't think I'd be the first to encounter this...I'll look later...

Alex313031, 2021-12-09:
@lurch @dmcardlenl Yup. chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox this has been an issue for as long as I have used etcher on linux. It will still run fine without making a symlink or copying /opt/balenaEtcher to /opt/balena-etcher-electron, but it is annoying nonetheless. It is probably an artifact from the days when the .debs were called and installed to "balena-etcher-electron". With how nitpicky (a good thing) balena is about fixing little things here and there in etcher, you would think they would have resolved this a long time ago.

bedawi, 2022-01-16:
I can confirm this error on Fedora 35 when version 1.7.3-1 is installed from RPM.

This link is broken: 
```/usr/bin/balena-etcher-electron -> /opt/balena-etcher-electron/balena-etcher-electron```

To fix the problem I run:
```bash
sudo rm /usr/bin/balena-etcher-electron
sudo ln -s /opt/balenaEtcher/balena-etcher-electron /usr/bin/balena-etcher-electron
```

raniesantos, 2022-04-25:
Also on Fedora 35, same issue. Thanks @bedawi for that quick fix.

dfunckt, 2023-07-12:
Dupe of #3486

closed 2023-07-12 by dfunckt