balena-io/etcher #3643: I/O error

kolbai, 2021-11-25:
- **Etcher version:** 1.7.0
- **Operating system and architecture:** Windows 10, 64bit
- **Image flashed:** any *.img file
- **Do you see any meaningful error information in the DevTools?** Yes.

I cannot flash any file to my SD card. I use an onboard Realtek RTS525A card reader. I get the following error:
```
description: "SDHC Card"
device: "\\.\PhysicalDrive2"
name: "Error"
message: "Couldn't clean the drive, Command failed: diskpart /s C:\Users\User\AppData\Local\Temp\etcher\tmp-dc4dfefa5631.tmp (code 2147942431)"
stack: "Error: Couldn't clean the drive, Command failed: diskpart /s C:\Users\User\AppData\Local\Temp\etcher\tmp-dc4dfefa5631.tmp (code 2147942431)
```

I found a suggestion to comment out the following line in _child-writer.js_ file:
```throw new Error('Couldn't clean the drive, ${error.message} (code ${error.code})');```

Once I do that, I get another error:
```
description: "SDHC Card"
device: "\\.\PhysicalDrive2"
errno: -4070
name: "Error"
syscall: "open"
message: "EIO: i/o error, open '\\.\PhysicalDrive2'"
stack: "Error: EIO: i/o error, open '\\.\PhysicalDrive2'"
```
What can I do to fix this issue?

kolbai, 2023-01-19:
Just an update. Issue remains unsolved in the latest version. Why it is happening is discussed here: https://github.com/pbatard/rufus/issues/2119. In short, etcher calls CreateFile function and passes `\\.\PhysicalDrive2` as a file name. For some reasons, Windows reports that device is not functioning and the whole process fails. Passing PNPDeviceID instead of DeviceID fixes this issue.

mcraa, 2023-01-19:
Thanks for your research, we will have a look.

evan-w-enovation, 2025-03-06:
I'm seeing this same error using belena-cli.
? Select a drive \\.\PhysicalDrive2 (30.9 GB) - 
? This will erase the selected drive. Are you sure? Yes
EIO: i/o error, open '\\.\PhysicalDrive2\'