balena-io/etcher #3645: A technical problem

hamadasal, 2021-11-29:
Can not complete downloading this application. it requires me a login and password after completing 80-88% of downloading and then stops. im using IDM on my browsers.. 
It takes until now about 5 times and i fail to fix this. please help!

hamadasal, 2021-11-29:
> Can not complete downloading this application. it requires me a login and password after completing 80-88% of downloading and then stops. im using IDM on my browsers.. It takes until now about 5 times and i fail to fix this. please help!

thundron, 2021-11-29:
That is super weird: where did you download Etcher from? We never ask for login/password anywhere; we just ask the password to enable the child-process to write to the selected targets.

hamadasal, 2021-11-30:
Yes.. so weird to me too.. IDM only asks me logins and passwords even after 80% completing the free downlaoding from the main open source website here: https://www.balena.io/etcher/ 

I have screenshoted the situation so that i want you share this hoping to get an explanation.
I dont use anothe rdownloading application than IDM with all my browsers.

the pictures belwo show what happened with me during 2 days while i start normally and after 60-80-88% of completing of my downlaod: 

(uploaded pictures in JPEG and PNG extentions)

[image]
[image]
[image]

[image]
[image]
[image]

hamadasal, 2021-11-30:
Hello there. i have replied your comment there. thank you.

lurch, 2021-11-30:
I wonder if "IDM" is trying to download the file in "multiple parts", and perhaps that's hitting some kind of API-rate-limit in GitHub? :shrug:

closed 2023-07-12 by dfunckt