balena-io/etcher #3646: Trouble installing the .deb on both Pop!_OS 21.04 & Zorin 16

the-questjon, 2021-11-29:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

user:~$ curl -1sLf \
>    'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
>    | sudo -E bash
[sudo] password:              
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=zorin  version=16  codename=focal  arch=x86_64  
 >>>>:
 NOPE: Checking for apt dependency 'apt-transport-https' ...
   OK: Updating apt repository metadata cache ...
   OK: Attempting to install 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Importing 'balena/etcher' repository GPG key into apt ...
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=zorin  version=16  codename=focal  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_ZR0XzjiXq (might not be errors, nor even relevant):
 !!!!:
 !!!!: > Warning: apt-key output should not be parsed (stdout is not a terminal)
 !!!!: > OK

the-questjon, 2021-11-29:
Please help. I would rather install this way and run natively. I don't want to run an AppImage.

mcraa, 2022-01-05:
see #3663

closed 2022-01-05 by mcraa