balena-io/etcher #3647: Etcher won't launch on Fedora 35.

offensivetester, 2021-12-03:
- **Etcher version: Latest (1.7.1)** 
- **Operating system and architecture: Fedora Linux 35 (Workstation Edition)** 
- **Image flashed: None** 

I tried running the AppImage, but every time I execute `./balenaEtcher-1.7.1-x64.AppImage` I get an error.
This is the log:
```
[Axios v1.7.1] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.1] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.1] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
{"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)\n    at TLSSocket.emit (events.js:315:20)\n    at TLSSocket._finishInit (_tls_wrap.js:932:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)","config":{"url":"https://balena.io/etcher/static/config.json","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.7.1"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"json","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1,"transitional":{"silentJSONParsing":true,"forcedJSONParsing":true,"clarifyTimeoutError":false}},"code":"CERT_HAS_EXPIRED"}
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[64615:1203/204326.427485:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenalHVK8D/balena-etcher-electron: line 10: 64615 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"
```

What do I do?

lurch, 2021-12-04:
Duplicate of #3639 ?

adatum, 2022-03-04:
Same problem with etcher v1.7.7 and Fedora 35

```
$ /opt/balenaEtcher/balena-etcher-electron
[Axios v1.7.7] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.7] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.7] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[286457:0304/150419.065587:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/opt/balenaEtcher/balena-etcher-electron: line 10: 286457 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"
```
This appears to be due to a bug in electron: https://github.com/electron/electron/pull/31091

It works if launched with the `--no-sandbox` or `--disable-gpu-sandbox` flag but [this is not advised according to electron's documentation](https://www.electronjs.org/docs/v14-x-y/tutorial/sandbox#disabling-chromiums-sandbox-testing-only).

leillo1975, 2022-04-07:
Have the same problem on Ubuntu Jammy (22.04)

kayD, 2022-07-19:
I have the same problem on Ubuntu Jammy (22.04) (German output)
```
{12:11}/opt/Backup/balena-etcher ➭ ./balenaEtcher-1.7.9-x64.AppImage 
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[41390:0719/121157.390911:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenauPPptM/balena-etcher-electron: Zeile 10: 41390 Trace/Breakpoint ausgelöst   (Speicherabzug geschrieben) "${script_dir}"/balena-etcher-electron.bin "$@"
```

theofficialgman, 2022-09-16:
@mcraa can you update etcher to electron >=14 or v13.5.0 (which has the change backported) instead of just marking issues

as you must be aware now, this is due to a GLIBC change on the host that required electron changes https://github.com/electron/electron/pull/31091 . etcher uses an ancient 12.X.Y electron version and this needs to be updated for multiple reasons (security fixes, wayland improvements in recent versions, etc)

mcraa, 2022-09-16:
@theofficialgman 
Yes, I am aware, thanks for the reminder anyway.
By v14 there are so many breaking changes that would mean an almost complete rewrite.
{spoiler alert} Then why not a more recent version like v18? {/spoiler alert}
But with that, you don't want to exclude any of the current users so it is a lot of checking to prove it works like before.

theofficialgman, 2022-09-16:
@mcraa and how about v13 then (v13.5.0+ which has the fix)? does this have any breaking changes that are critical to function?

yes I knew that one of the electron versions fundamentally changed how it worked (though it was 15 but I guess its 14) but a rewrite needs to be done at some point. Software which can't run an modern systems isn't helpful.

theofficialgman, 2022-09-16:
> But with that, you don't want to exclude any of the current users so it is a lot of checking to prove it works like before.

electron 19 has the same glibc minimum requirements as v12 across all linux platforms and great wayland/xorg compatibility. the external dependencies are still the same

mcraa, 2022-09-16:
When bumping the version needs quite some effort and it will need the same effort in a short time, I think focusing on the rewrite with a more future-proof version makes sense. It is a pity that from the outside it looks like nothing is happening for a long time.

theofficialgman, 2022-09-16:
> When bumping the version needs quite some effort and it will need the same effort in a short time, I think focusing on the rewrite with a more future-proof version makes sense. It is a pity that from the outside it looks like nothing is happening for a long time.

this is just untrue and you know it. you have had over a year to deal with this issue and nothing has been done at all. all you do is bump the version from 12.x.y to ^13.5.0 and build. I have already checked v13 and there don't appear to be any breaking api changes that you depend on in it. please list any api changes you are aware of that prevent this simple change

theofficialgman, 2022-11-13:
@mcraa still no fix? like I said. litearlly you need zero changes. bump to electron 13.5.0+ and spectron to 15.0.0 for testing. it works fully without this issue. I have built and tested locally with no problems.

adatum, 2022-11-13:
@theofficialgman Since you've already tested locally, maybe you could make a pull request?

theofficialgman, 2022-11-14:
> @theofficialgman Since you've already tested locally, maybe you could make a pull request?

a PR is not what the devs want. they don't want to update, that is clear. For some reason they are opposed to doing anything. They would rather the bug reports pile up. They said it would be "quite some effort" and that it would "need a lot of checking to prove it works as before"

lurch, 2022-11-15:
> For some reason they are opposed to doing anything.

I wouldn't go that far - there's been a whole _heap_ of updates recently? https://github.com/balena-io/etcher/blob/master/CHANGELOG.md
(although I dunno if any of them fix the original issue mentioned?)

theofficialgman, 2022-11-15:
> > For some reason they are opposed to doing anything.
> 
> I wouldn't go that far - there's been a whole _heap_ of updates recently? https://github.com/balena-io/etcher/blob/master/CHANGELOG.md (although I dunno if any of them fix the original issue mentioned?)

Yes I can go that farm there were no commits for over 6 months. All the commits from the last week are not merges simply bumping the version of dependencies. Except for one commit that changes how webpack is used to build the application, there haven't been any real changes.
Nothing changed to affect this bug either.

ThatOhio, 2022-12-01:
Just chiming in to say that this same error happens on Pop!_OS.

pavgup, 2023-01-13:
Found a solution by running with `/opt/balenaEtcher/balena-etcher-electron --in-process-gpu` -- not entirely certain if this is the right move.

mcraa, 2023-01-16:
@pavgup could you try v1.13.0 and above, should not need to add any args then.

adatum, 2023-01-16:
> @pavgup could you try v1.13.0 and above, should not need to add any args then.

Resolved at last! v13.3 launches properly. 

I had to first uninstall `balena-etcher-electron` as the old package was named, otherwise the installation of `balena-etcher` failed due to conflicts.

mcraa, 2023-01-16:
> > @pavgup could you try v1.13.0 and above, should not need to add any args then.
> 
> I had to first uninstall `balena-etcher-electron` as the old package was named, otherwise the installation of `balena-etcher` failed due to conflicts.

Most package managers warn you about that, but thanks for pointing it out.
The name change made it from the command line launch work more or in more cases out of the box.

closed 2023-01-16 by mcraa

theofficialgman, 2023-01-16:
over a year late and 4 months that I said there were no issues... its about time you make the minor version bump