balena-io/etcher #3648: Cannot read property ‘message’ of null

eleius, 2021-12-06:
- **Etcher version:** 1.7.1
- **Operating system and architecture:** Arch Linux x86_64
- **Image flashed:** [latest deb](https://github.com/balena-io/etcher/releases/download/v1.7.1/balena-etcher-electron_1.7.1_amd64.deb) via AUR package
- **Do you see any meaningful error information in the DevTools?** 

As reported by users on the [forum](https://forums.balena.io/t/flash-failed-cannot-read-property/350420), verification after flashing ISO files always fails with message: “Cannot read property ‘message’ of null.”

CraigGoodspeed, 2021-12-08:
Same result on Ubuntu 21.10

CraigGoodspeed, 2021-12-08:
So I resolved the issue by following the installation instructions on

https://github.com/balena-io/etcher?d_id=6f49fb31-ffa2-4dcc-a79a-04fbcf298f0fR#debian-and-ubuntu-based-package-repository-gnulinux-x86x64

Then using the installed application.

It would fail when downloading the AppImage from https://www.balena.io/etcher/ this would cause the error.

Hope this helps someone out there.

Alex313031, 2021-12-09:
@CraigGoodspeed I just installed the 1.7.1 .deb and it still does this on debian bullseye @se4n01 @eleius

Alex313031, 2021-12-09:
It also has a weird thing where when it's writing it will dip down to .30 mb/s for like 10 seconds and then resume at normal speed.

GitYing, 2021-12-15:
> @CraigGoodspeed I just installed the 1.7.1 .deb and it still does this on debian bullseye @se4n01 @eleius

Download the 1.7.0 version and everything works perfectly fine again

qwertychouskie, 2021-12-26:
Issue affects my system too.

athaarvaaagit, 2021-12-27:
I using the v.1.72 and still it is giving this erro
[image]
r

athaarvaaagit, 2021-12-27:
please check

Jeffrey-H, 2021-12-27:
using 1.7.2 on Fedora 34. Same issue.

dallinbriggs, 2021-12-27:
> So I resolved the issue by following the installation instructions on [...]

This did not work for me. I tried the installed version and the app image of 1.7.2 and they both gave me the same error. 1.7.0 indeed fixed my issue.

closed 2021-12-29 by bulldozer-balena[bot]

SamuelBanya, 2022-01-10:
You shouldn't close a GitHub issue without at least linking to a version that actually works.

Please do this before auto-closing issues, Balena Etcher team.

For anyone looking for the 'AppImage' version, the releases can be found here:
https://github.com/balena-io/etcher/releases 

Just use Version 1.7.0 App image and it will work fine on Manjaro Linux:
https://github.com/balena-io/etcher/releases

ZaLiTHkA, 2022-01-10:
@SamuelBanya, looking through the list of releases, version `1.7.3` was tagged at "Dec 29, 2021, 5:10 PM GMT+2" with only one commit *specifically* for this issue..

besides, there have been two releases since version `1.7.1`, so why would you recommend people use `1.7.0`? 🤔

SamuelBanya, 2022-01-10:
@ZaLiTHkA 
This is because unless you use that Version 1.7.0, you will experience the OP's issue of 'Cannot read property ‘message’ of null' after attempting to flash an ISO with Balena Etcher.

The easier solution is just to use the older App Image until the team actually fixes this issue on Manjaro (or any Arch based distro).

PR #3667 (2021-12-29): patch: fix mesage of null
resolves #3648

Change-type: patch