balena-io/etcher #3650: Fake GitHub profile for Balena?

Technetium1, 2021-12-07:
Is the user [balenaEtcher](https://github.com/balenaEtcher) an official representative of Balena? If not they should probably be reported.

lurch, 2021-12-07:
Looking at the sourcecode for the `README.md` on there, they're also hosting content on `https://balenaetcher.eu` (which might be similarly "fake"?)

Alex313031, 2021-12-09:
@Technetium1 @lurch Made this issue over there > https://github.com/balenaEtcher/balenaEtcher/issues/1#issue-1075096380

lurch, 2021-12-09:
The other thing that points to it being "fishy" :fish: is that the other repo lists the latest version as v1.5.113 whereas this official repo lists the latest version as v1.7.1

Alex313031, 2021-12-10:
@lurch well it hasn't been updated in a long time, and the time coincides with the time the actual v1.5.113 was released so that doesn't necessarily add to the fishiness.

Alex313031, 2021-12-10:
@lurch I'm going to download the official source code for v1.5.113 and this repos v.1.5.113 in a virtual machine and do a diff to see what if any differences there are. When I'm done I will post here if I find anything malicious.

20k-ultra, 2021-12-20:
Hey everyone, really appreciate the diligence here. That org has no relations to Balena or the Etcher project. In fact, we have analyzed the code hosted at their domain balenaetcher.eu so we could provide proof that it contained malicious code and attempt to get it taken down (of which we did).

We will look into what we can do about this org to avoid any confusion.

closed 2021-12-20 by 20k-ultra

Alex313031, 2021-12-21:
@20k-ultra Is this why that domain shows a 403? When I looked at the code on their github, I went to that domain and it was already taken down. Did you guys get it taken down before this issue was created?

20k-ultra, 2021-12-21:
I'm not sure when we got it taken down but I remember hearing about that domain a few months ago so it's possible @Alex313031

lurch, 2021-12-22:
The index-page of the domain 403s, but `https://balenaetcher.eu/wp-content/uploads/2020/12/balenaetcher-do-pobrania-za-darmo-s.jpeg` still works (which is the image used in their `README.md`). I.e. the domain itself hasn't been "taken down" (yet).

Technetium1, 2022-02-09:
Update: The profile was removed by GitHub.