balena-io/etcher #3652: Cannot read property 'message' of null popup during finishing process

JOmarCuenca, 2021-12-10:
- **Etcher version:** Version = 1.7.1
- **Operating system and architecture:** Linux 5.4.0-91-generic Ubuntu SMP Fri Nov 5 16:31:28 UTC 2021 x86_64 GNU/Linux
- **Image flashed:** pop-os_20.04_amd64_nvidia_35.iso

Hi, there seems to be some sort of bug during the final process.
The live usb seems to have been created successfully (pending to be tested), but after the finishing process a message pops up saying **Cannot read property 'message' of null**.

```
gui.js:35 Verifying 1 device, 100% at 23.20 MB/s (total 23.20 MB/s) eta in 0s with 0 failed devices
gui.js:46 Finish: 3038543872
gui.js:46 error null
gui.js:46 Terminating IPC server
gui.js:46 Flash results Object
gui.js:46 Fri Dec 10 2021 13:23:59 GMT-0600 (Central Standard Time) Write failed ({"image":{"size":3038543872,"name":"pop-os_20.04_amd64_nvidia_35.iso","hasMBR":true,"partitions":[{"offset":247808,"size":4194304,"type":239,"index":1}],"extension":"iso","path":"/home/omar/distros/pop-os_20.04_amd64_nvidia_35.iso"},"drives":[{"size":16001269760,"isVirtual":false,"enumerator":"lsblk:json","logicalBlockSize":512,"raw":"/dev/sda","error":null,"isReadOnly":false,"displayName":"/dev/sda","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/sda","busVersion":null,"isSystem":false,"busType":"USB","isCard":null,"isUSB":true,"devicePath":"/dev/disk/by-path/pci-0000:00:14.0-usb-0:2:1.0-scsi-0:0:0:0","mountpoints":[{"path":"/media/omar/KINGSTON","label":"KINGSTON"}],"description":"Kingston DT_101_G2 (KINGSTON)","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"e8b768cd-37ce-45bb-8e55-d3b87d906e0e","status":"failed","flashInstanceUuid":"e8b768cd-37ce-45bb-8e55-d3b87d906e0e","error":{},"sample":0.1,"applicationSessionUuid":"d978a5de-5c24-4fe7-b321-7f131195d888","flashingWorkflowUuid":"aba82f0c-eb71-46e8-be6e-c04244bd8796"})
gui.js:46 {"stack":"TypeError: Cannot read property 'message' of null\n    at file:///tmp/.mount_balenanUvM8E/resources/app/generated/gui.js:351:4728\n    at async withTmpFile (file:///tmp/.mount_balenanUvM8E/resources/app/generated/gui.js:1:344288)\n    at async elevateCommand
 (file:///tmp/.mount_balenanUvM8E/resources/app/generated/gui.js:351:4236)\n    at async Server.<anonymous> (file:///tmp/.mount_balenanUvM8E/resources/app/generated/gui.js:351:8340)","message":"Cannot read property 'message' of null","image":"pop-os_20.04_amd64_nvidia_35.iso"}
console.<computed> @ gui.js:46
```
The lines above are the last part of a log file that I generated from the console.
I also attached the original log file with this issue. [popOS_etcher_failed_on_finish.log](https://github.com/balena-io/etcher/files/7695133/popOS_etcher_failed_on_finish.log)

lurch, 2021-12-10:
Duplicate of #3648 ?

JOmarCuenca, 2021-12-15:
> Duplicate of #3648 ?

Yes, it seems to be the same issue.

closed 2021-12-15 by JOmarCuenca