balena-io/etcher #3653: Incorrect SHA512 checksum in latest.yml

allywilson, 2021-12-12:
- **Etcher version: 1.7.1** 
- **Operating system and architecture: NA** 
- **Image flashed: NA** 
- **Do you see any meaningful error information in the DevTools?** 

**latest.yml** file contains incorrect SHA512 checksum for the latest **balenaEtcher-Setup-1.7.1.exe** download.

Contains: **byitThHrH8eTAwIGVi9IMxBEtfJe+tO4XOE9eREds6S12GiWuIA9TFNFsQg1yEsoDbNdWKKXd2dF8Ug8394RUQ==**
Download is: **6f28ad4e11eb1fc793030206562f48331044b5f25efad3b85ce13d79111db3a4b5d86896b8803d4c5345b10835c84b280db35d58a297776745f1483cdfde1151**

It doesn't actually look like a SHA512 checksum, it looks like something has been base64 encoded, but I got gibberish when decoding it.

lurch, 2021-12-12:
See also #2826

allywilson, 2021-12-12:
2 years and not fixed or addressed? Why?

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)
closed 2023-12-21 by dfunckt