balena-io/etcher #3656: url encoded flash link doesn't work

AskAlice, 2021-12-16:
- **Etcher version:** 1.7.1
- **Operating system and architecture:** Win 11 amd64
- **Image flashed:** balena sound
- **Do you see any meaningful error information in the DevTools?** 

[image]
I thought it might have been the unicode in the SSID but I tried removing it and I'm getting the same issue. the error contained my authorization token so i redacted it

```
gui.js:formatted:58144 {"stack":"Error: ENOENT: no such file or directory, open 'C:\\Windows\\system32\\%7B%22method%22%3A%22POST%22%2C%22url%22%3A%22https%3A%2F%2Fapi.balena-cloud.com%2Fdownload%3FappId%3D1884746%26deviceType%3Draspberrypi4-64%26version%3D2.88.4.prod%26network%3Dwifi%26wifiSsid%3D0_0%20(%20%CD%A1%C2%B0%20%CD%9C%CA%96%20%CD%A1%C2%B0)%26appUpdatePollInterval%3D10%22%2C%22headers%22%3A%7B%22Authorization%22%3A%22Bearer%20redacted%22%7D%2C%22data%22%3A%7B%22appId%22%3A1884746%2C%22deviceType%22%3A%22raspberrypi4-64%22%2C%22version%22%3A%222.88.4.prod%22%2C%22downloadConfigOnly%22%3Afalse%2C%22network%22%3A%22wifi%22%2C%22wifiSsid%22%3A%220_0%20(%20%CD%A1%C2%B0%20%CD%9C%CA%96%20%CD%A1%C2%B0)%22%2C%22wifiKey%22%3A%22%22%2C%22appUpdatePollInterval%22%3A10%7D%7D'","message":"ENOENT: no such file or directory, open 'C:\\Windows\\system32\\%7B%22method%22%3A%22POST%22%2C%22url%22%3A%22https%3A%2F%2Fapi.balena-cloud.com%2Fdownload%3FappId%3D1884746%26deviceType%3Draspberrypi4-64%26version%3D2.88.4.prod%26network%3Dwifedactedi%26wifiSsid%3D0_0%20(%20%CD%A1%C2%B0%20%CD%9C%CA%96%20%CD%A1%C2%B0)%26appUpdatePollInterval%3D10%22%2C%22headers%22%3A%7B%22Authorization%22%3A%22Bearer%r%22%7D%2C%22data%22%3A%7B%22appId%22%3A1884746%2C%22deviceType%22%3A%22raspberrypi4-64%22%2C%22version%22%3A%222.88.4.prod%22%2C%22downloadConfigOnly%22%3Afalse%2C%22network%22%3A%22wifi%22%2C%22wifiSsid%22%3A%220_0%20(%20%CD%A1%C2%B0%20%CD%9C%CA%96%20%CD%A1%C2%B0)%22%2C%22wifiKey%22%3A%22%22%2C%22appUpdatePollInterval%22%3A10%7D%7D'","errno":-4058,"code":"EUNPLUGGED","syscall":"open","path":"%7B%22method%22%3A%22POST%22%2C%22url%22%3A%22https%3A%2F%2Fapi.balena-cloud.com%2Fdownload%3FappId%3D1884746%26deviceType%3Draspberrypi4-64%26version%3D2.88.4.prod%26network%3Dwifi%26wifiSsid%3D0_0%20(%20%CD%A1%C2%B0%20%CD%9C%CA%96%20%CD%A1%C2%B0)%26appUpdatePollInterval%3D10%22%2C%22headers%22%3A%7B%22Authorization%22%3A%22Bearer%redacted%22%7D%2C%22data%22%3A%7B%22appId%22%3A1884746%2C%22deviceType%22%3A%22raspberrypi4-64%22%2C%22version%22%3A%222.88.4.prod%22%2C%22downloadConfigOnly%22%3Afalse%2C%22network%22%3A%22wifi%22%2C%22wifiSsid%22%3A%220_0%20(%20%CD%A1%C2%B0%20%CD%9C%CA%96%20%CD%A1%C2%B0)%22%2C%22wifiKey%22%3A%22%22%2C%22appUpdatePollInterval%22%3A10%7D%7D"}```

mcraa, 2021-12-22:
Thanks @AskAlice for sharing. 
Does 1.7.2 make this problem disappear for you?

ps.: nice wifi ssid

closed 2022-01-03 by mcraa

AskAlice, 2022-08-06:
@mcraa no it's still there
```
https://www.balena.io/etcher/open-image-url/?imageUrl=%257B%2522method%2522%253A%2522POST%2522%252C%2522url%2522%253A%2522https%253A%252F%252Fapi.balena-cloud.com%252Fdownload%253FappId%253D1864196%2526releaseId%253D2116033%2526deviceType%253Draspberrypi3%2526version%253D2.98.33%2526network%253Dwifi%2526wifiSsid%253D0_0%2520(%2520%25CD%25A1%25C2%25B0%2520%25CD%259C%25CA%2596%2520%25CD%25A1%25C2%25B0)%2526wifiKey%253Di%2520just%2520lost%2520the%2520game%2526appUpdatePollInterval%253D10%2522%252C%2522data%2522%253A%257B%2522appId%2522%253A1864196%252C%2522releaseId%2522%253A2116033%252C%2522deviceType%2522%253A%2522raspberrypi3%2522%252C%2522version%2522%253A%25222.98.33%2522%252C%2522network%2522%253A%2522wifi%2522%252C%2522wifiSsid%2522%253A%25220_0%2520(%2520%25CD%25A1%25C2%25B0%2520%25CD%259C%25CA%2596%2520%25CD%25A1%25C2%25B0)%2522%252C%2522wifiKey%2522%253A%2522i%2520just%2520lost%2520the%2520game%2522%252C%2522appUpdatePollInterval%2522%253A10%257D%257D
```

AskAlice, 2022-08-06:
you could use something like this to encode. Many strings will actually end up less characters than they started off with, but of course it's utf-8 so it will use 4 more bits per character for most characters
https://stackoverflow.com/a/71021962/643875

AskAlice, 2022-08-06:
to clarify, my error is different now, but it's similar enough to be in this same issue. 
[image]

mcraa, 2022-08-08:
Thanks for coming back.
I checked your open in etcher URL, and removing the wifi config from it made it work.
So it confirms, we need to find a way to deal with them.

Until then you can click the down arrow at the end of the `Flash` button on the dashboard and select download, then open the download image like it was with earlier versions of etcher.

AskAlice, 2022-08-12:
> Thanks for coming back. I checked your open in etcher URL, and removing the wifi config from it made it work. So it confirms, we need to find a way to deal with them.
>
> Until then you can click the [...]

If that encoding method scares you, another thing you could do is first have a link that opens etcher via a URI with some basic info about the image, then once that opens it runs an http server and awaits post input from the same page, then closes the http server once done.