balena-io/etcher #3658: Why is Etcher 139 MB large?

rillig, 2021-12-17:
- **Etcher version:** 1.7.1 
- **Operating system and architecture:** Windows 10 amd64

When I downloaded Etcher, I was negatively surprised by the size of the download. For a tool that copies a file to a disk device, I would have expected 5 MB at most. On Unix-like operating systems, a simple `dd if=image out=device` is enough, so it shouldn't be necessary that the Windows equivalent is a million times larger.

Do you have plans to reduce the file size of the distributed binaries?

lurch, 2021-12-18:
See #2974 ?

A9G-Data-Droid, 2022-01-06:
After pointing to a closed issue that points to another closed issue. The answer seems to be "framework". It's Electron. Let's face it, when you're developing a desktop app with web technology then you gotta think that there is a web-server and web-browser inside each application. I'm not sure why this became popular, it's a terrible idea. 

There are plenty of good ways to develop cross platform apps these days. Rust being first choice and even .NET6 would be a better choice than Electron!

ZaLiTHkA, 2022-01-07:
hey @A9G-Data-Droid,

> There are plenty of good ways to develop cross platform apps these days. Rust being first choice and even .NET6 would be a better choice than Electron!

found a comment that may explain why this project is _still_ using Electron at this point in time:

> we were considering using other frameworks, but given the massive amount of electron APIs we use right now it's something that will probably come in the far, far future. closing for now, will re-open accordingly when we get our hands on it
>
> _https://github.com/balena-io/etcher/issues/3568#issuecomment-910199560 by @thundron_

so perhaps there's simply too much that would need to change? I haven't dug through the code myself (yet?), might do that at some point and see what comes of it.

thundron, 2022-01-08:
@ZaLiTHkA correct! hopefully Electron will be more performant in the future, as there's multiple reasons why it still is a choice even at this point in time other than just legacy :) one example being exactly using web technologies which are very widespread!

xdpirate, 2022-01-09:
Why not just use Launchpad, like almost every other software project does for its PPA...? Still getting this error message.

A9G-Data-Droid, 2022-01-10:
@thundron I think the use case for Electron is when you want to make a web application and also offer a local version using the same code base. It then makes sense to reuse the code you already had. If an application is local only, like an image burning utility, then it doesn't really make sense to code it this way. Does it need to be hosted on the web? Does it need to be accessed remotely?

Maybe you could make an image burning server but someone with physical access to the hardware is needed to connect and remove physical devices. It's a strange design decision.

lurch, 2022-01-10:
> If an application is local only, like an image burning utility, then it doesn't really make sense to code it this way.

I think it's because Balena is (or at least was) pretty deeply embedded in the Javascript / NodeJS ecosystem, and so it made sense when Etcher was first being developed? :shrug:

nerdbaggy, 2022-01-15:
@A9G-Data-Droid 

> There are plenty of good ways to develop cross platform apps these days. Rust being first choice and even .NET6 would be a better choice than Electron!

Not even close. If you want to create a cross platform GUI electron is the way for now

No good gui for rust - https://www.areweguiyet.com/
.net is currently in preview for cross platform gui - https://docs.microsoft.com/en-us/dotnet/maui/what-is-maui

ZaLiTHkA, 2022-01-16:
FWIW, NodeGUI does look like a really good alternative (on paper, at least), but unfortunately [an issue with broken hot-reloading](https://github.com/nodegui/react-nodegui/issues/354) makes it really difficult to evaluate _properly_.. 🤔

Fatih120, 2022-01-18:
I really hate the webapp trend; I do not want to live in a future where all my applications that should be 2mb applets are each their own web browser especially when here I can just fall back onto rufus. Don't get me wrong, etcher works well, but choices like that add to the bloated internet problem.

anzenchitai, 2023-02-02:
I decided to give it a try after facing some issues with Rufus. 125 megabytes just to burn an image to a USB flash drive? Just no.

davidak, 2023-02-02:
@anzenchitai try [USBImager](https://bztsrc.gitlab.io/usbimager/) instead. It says "less than 192 Kilobytes".

My1, 2023-08-10:
> I really hate the webapp trend; I do not want to live in a future where all my applications that should be 2mb applets are each their own web browser especially when here I can just fall back onto [...]

not only that these monsters obviously also consume a lot more working resources, i.e. ram and CPU

thynus, 2024-11-24:
On my m1 mac mini the app is 365.9MB.  Meanwhile, Audacity which uses wxWidgets is only 169MB.  However the enormous binary size may be a non-issue for most consumers and require an investment in developer time that would not generate revenue?

My1, 2024-11-24:
on Mac the issue is a lot worse than on other platforms as on Mac you have (once again) the Universal Binary which packages the application for 2 architectures in one package, while currently needed so the application can be more easily pushed to both Intel and Apple Silicon Macs, that thing was also used during the transition away from PowerPC to Intel, as well as for the transition from 32 to 64 bit on Intel.

thynus, 2024-11-24:
Unless I am missing something this is the Apple Silicon only binary not a Universal Binary according to the get info box.  FYI Universal Binaries out of the box packages only 2 different platform binaries (YMMV) the brilliant emacsformacos packages more than 2 in their Universal Binary.

thynus, 2024-11-24:
Found this nice write up:
https://www.umlboard.com/blog/moving-from-electron-to-tauri-1/

jcormier, 2025-01-08:
> @anzenchitai try [USBImager](https://bztsrc.gitlab.io/usbimager/) instead. It says "less than 192 Kilobytes".

Wow, I was excited to hear about a different cross platform disk imager software.  And I have to say I am very disappointed in USBImager.  The GUI is so minimal that it is not user-friendly and developer consistently closes any issues bringing improvements to usability (as is his right).  Etcher has things laid out so well that it's immediately obvious to anyone what steps 1/2/3 are, it's minimal but clear.

Other missing features:
* Doesn't estimate disk speed
* No bmap support
* No ext4 speed improvements. (This is a big feature of etcher)
* Doesn't validate image was written correctly
* Doesn't support checking md5sum
* Doesn't support writing multiple sdcards at once

lurch, 2025-01-09:
> Wow, I was excited to hear about a different cross platform disk imager software.

There's also [Raspberry Pi Imager](https://www.raspberrypi.com/software/). Whilst it's obviously primarily targeted at Raspberry Pi users (and so doesn't have all the same features as Balena Etcher), you can still use the "Use custom" option to write any OS image to any USB drive.

lurch, 2025-01-09:
I obviously don't know what your planned usage is, but note that you _can_ supply your own images JSON file to RPi Imager, if you want it to display a different set of images. See https://github.com/raspberrypi/rpi-imager/?tab=readme-ov-file#custom-repository (with https://downloads.raspberrypi.com/os_list_imagingutility_v4.json being the default images URL).

Balena-folks: Apologies for the slightly off-topic discussion
@jcormier: If you have any further questions, please ask at https://github.com/raspberrypi/rpi-imager/issues

jcormier, 2025-01-09:
Thanks I'll check it out

KubaO, 2025-03-19:
`we were considering using other frameworks, but given the massive amount of electron APIs we use right now` - for a disk imager? Massive amount? This is like a "Hello World" but slightly bigger thing for WIN32 API. It'd use about 100 functions from WinAPI, the whole thing, and for anyone who does it for a living would take a week to write, another week to get tests and CI set up. Electron is a scourge on this planet. Just imagine how much traffic is consumed by all the apps people download, each one of them including some version of it because everyone and their dog uses it now.

thynus, 2025-04-11:
Demo on how to quickly migrate from Electron to Tauri, could Balena use this to move off electron?

https://github.com/simonhyll/electron-to-tauri
https://www.youtube.com/watch?v=fMTtMRLUnIY

thynus, 2026-09-08:
Maybe AI could be used to create a Python script to port the whole thing to Qt?

KeithIMyers, 2026-09-09:
> Maybe AI could be used to create a Python script to port the whole thing to Qt?

There are already several things like that already