balena-io/etcher #3660: Analytics tracking is broken

pranasziaukas, 2021-12-21:
Analytics tracking events are not reaching their destinations AFAICT.

> it's a bug that was just recently solved in Electron

As I understand, there was a recent issue with the upstream Electron library.

Etcher is dependent on Electron [v12.0.2](https://github.com/balena-io/etcher/blob/b938132038729c05be7e0db2ea150377ac3fa87f/package-lock.json#L6091), which was released [back in March](https://github.com/electron/electron/releases/tag/v12.0.2).

> it's about the change in the Let's Encrypt certificates that happened 1st October 2021

I think I found the relevant fix included in v12.2.1 - https://github.com/electron/electron/releases/tag/v12.2.1 - the timing matches.

Internal discussion [here](https://www.flowdock.com/app/rulemotion/r-thefunnel/threads/bV7BGwvdme0qtehSgo7SCIv0_LJ).

---

TL;DR please bump the Electron dependency to at least v12.2.1

mcraa, 2022-02-21:
@pranasziaukas fixed with #3704

closed 2022-02-21 by mcraa

pranasziaukas, 2022-02-21:
Thanks @mcraa. Just to make sure, will users have to update to the latest version of balenaEtcher or is the fix done server-side?

mcraa, 2022-02-21:
@pranasziaukas, it is client-side, they need to update.
The files are here https://github.com/balena-io/etcher/releases/tag/v1.7.6
etcher.io is not populated correctly at the moment, shows only 1.7.5