balena-io/etcher #3663: Add support for Pop!_OS 21.10

Haprog, 2021-12-22:
Please add support for `distro=pop  version=21.10  codename=impish  arch=x86_64 `

This worked with the previous version 21.04 but I just upgraded to 21.10.

Running the install script gives me the following error.

```
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=pop  version=21.10  codename=impish  arch=x86_64  
 >>>>:
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Importing 'balena/etcher' repository GPG key into apt ...
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=pop  version=21.10  codename=impish  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_gRScLuVLA (might not be errors, nor even relevant):
 !!!!:
 !!!!: > Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d
 !!!!: > instead (see apt-key(8)).
 !!!!: > OK

```

mcraa, 2022-01-05:
AFAIK Pop is based on ubuntu, probably they updated the distro name environment variable with the last release.
As you see in the logs you can force the setup to use a specific distro name in this case you will may use ubuntu/focal.
Same instructions detailed in the [cloudsmith setup guide](https://cloudsmith.io/~balena/repos/etcher/setup/#formats-deb)

closed 2022-01-05 by mcraa

Haprog, 2022-01-10:
The issue is that the install script should work on latest stable Ubuntu and/or Pop!_OS without having to manually override distro/codename/version to an older release to make it install. Is there another ticket for this?

This was not an issue on Pop!_OS 21.04 which is based on Ubuntu 21.04 (Hirsute Hippo), but it is an issue on 21.10 (Impish Indri) so there is clearly an issue that the install script does not support or recognize Impish Indri.

I'm not sure if the script differentiates between Ubuntu and Pop!_OS or is it just a matter of version/codename (which are the same for Ubuntu and Pop!_OS).

I could try to override the version to 21.04 hirsute to see if it works but as a user trying to install and getting this error for the first time I have no way of knowing if I can expect the app to install or work properly if I need to override to use an older release version since there may be significant changes between the OS versions.

aindriu80, 2023-03-16:
doesn't work on Pop 22.04

`curl -1sLf \                                                                                                                                                                                                       100 ✘  at 18:41:56   ▓▒░
  'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
  | sudo -E distro=pop codename=jammy arch=x86_64 bash
`