balena-io/etcher #3665: Unable to install balena etcher through the debian & ubuntu repository method

sylindrical, 2021-12-24:
- **Etcher version:**    1.7.2
- **Operating system and architecture:** Zorin Os 16
- **Image flashed:**  ?
- **Do you see any meaningful error information in the DevTools?** 

To install as balena etcher through this way(https://github.com/balena-io/etcher?d_id=6f49fb31-ffa2-4dcc-a79a-04fbcf298f0fR#debian-and-ubuntu-based-package-repository-gnulinux-x86x64) in the Debian and Ubuntu based repositories, i tried the first command. And i got an error saying

Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.

Before it said that 8 lines or so before, it gave this error:

 NOPE: Checking for apt dependency 'apt-transport-https' ...
 NOPE: Updating apt repository metadata cache ...
Failed to update via apt-get update

lurch, 2021-12-25:
`Failed to update via apt-get update`

Does a plain `sudo apt-get update` work?

dsgsdg3, 2021-12-25:
download
https://github.com/balena-io/etcher/releases/tag/v1.7.0
balena-etcher-electron-1.7.0-linux-x64.zip

prince-of-hackers, 2021-12-31:
Trying to install on Debian today, got an error "402 Payment Required" on the first step from the Cloudsmith hosting server.

jmknoble, 2022-01-02:
Me Too^TM:

```
$ sudo apt update
[...]
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/focal/InRelease  402  Payment Required [IP: 2600:9000:2196:de00:e:f4d2:20c0:93a1 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu focal InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
[...]
```

jmknoble, 2022-01-02:
See also #3511

prince-of-hackers, 2022-01-02:
So apparently the repo has a 200gb per 30 days limit.

mcraa, 2022-01-05:
about the cloudsmith part see #3672

on the original issue: the two `NOPE` at the end of the report shows that it is a problem with apt not etcher, [lurch's suggestion](https://github.com/balena-io/etcher/issues/3665#issuecomment-1000949059) could help, or using the `fix-missing` option with apt, depending on your OS/setup

closed 2022-01-05 by mcraa