balena-io/etcher #3669: Ruined another USB

C0DE-L00P, 2021-12-29:
you know what? when you ruin my first usb stick I said no this must be me doing something wrong but 2 usb sticks in a row failed to flash and ended up to be in no write access mode with no hope to reformat it and not even get detected

I have faced this issue before with different usb stick but I give you another shot cuz I put my trust in your debugging skills and now is 1.7 version that issue must been solved now, and guess what ...

lurch, 2021-12-29:
https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/ ?

mcraa, 2021-12-30:
or https://itsfoss.com/format-live-linux-usb/ ?

I hope some of the links help, but as none of the information that is suggested in the issue template is provided, we can't really do anything with it.

closed 2021-12-30 by mcraa