balena-io/etcher #3670: SD card clone fails with "EBUSY: resource busy or locked"

martinkrulltott, 2021-12-29:
- **Etcher version: 1.7.1** 
- **Operating system and architecture: Mac OS 12.0, M1 Mac (ARM)** 
- **Image flashed: Raspberry Pi SD card** 
- **Do you see any meaningful error information in the DevTools?** 
```
Flashing 1 device, 44% at 21.90 MB/s (total 21.90 MB/s) eta in 809s with 0 failed devices
gui.js:35 Flashing 1 device, 44% at 21.40 MB/s (total 21.40 MB/s) eta in 828s with 0 failed devices
gui.js:46 Wed Dec 29 2021 21:09:20 GMT+0100 (Central European Standard Time) Write failed ({"image":{"path":"/dev/disk4","displayName":"/dev/disk4","description":"/dev/disk4","size":31914983424,"drive":{"size":31914983424,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk4","error":null,"isReadOnly":false,"displayName":"/dev/disk4","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk4","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/arm-io/usb-drd2@2280000/usb-drd2-port-hs@02100000","mountpoints":[{"path":"/Volumes/boot","label":"boot"}],"description":"Generic MassStorageClass Media","isUAS":null,"partitionTableType":"mbr"}},"drives":[{"size":31914983424,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk6","error":null,"isReadOnly":false,"displayName":"/dev/disk6","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk6","busVersion":null,"isSystem":false,"busType":"Secure Digital","isCard":true,"isUSB":false,"devicePath":"IODeviceTree:/arm-io/apcie@90000000/pci-bridge1@1/pcie-sdreader@0","mountpoints":[],"description":"Apple SDXC Reader Media","isUAS":null,"partitionTableType":"gpt"}],"driveCount":1,"uuid":"19548f98-3623-4c40-8417-41b48a042503","status":"failed","flashInstanceUuid":"19548f98-3623-4c40-8417-41b48a042503","error":{"name":"Error","code":"EBUSY","syscall":"read","errno":-16},"sample":0.1,"applicationSessionUuid":"7d97e18b-76bb-4cd2-82f2-9985bd0e073b","flashingWorkflowUuid":"0bf2aaa7-498a-4b12-b861-c859ddaab738"})
gui.js:46 {"stack":"Error: EBUSY: resource busy or locked, read","message":"EBUSY: resource busy or locked, read","name":"Error","code":"EBUSY","syscall":"read","errno":-16,"image":"disk4"}
console.<computed> @ gui.js:46
logException @ gui.js:46
flashImageToDrive @ gui.js:351
gui.js:46 Terminating IPC server
gui.js:46 Flash results {cancelled: false, skip: false}
gui.js:46 POST https://o95242.ingest.sentry.io/api/6075635/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=d7bc1f9e943b463f8ddf3c5ec437e50d net::ERR_CERT_INVALID
```

I'm trying to clone the SD card (32gb) in my Raspberry Pi to another card. I've got a brand new USB card reader and a brand new card for the target.
The process says its supposed to take about 15min, but 10min in it fails and shows an error message `EBUSY: resource busy or locked`. I attached the log output above.
I tried running Etcher as `sudo` and I've tried by first reformatting the target card to either FAT32 or HFS+, no luck.

Generally Im having some issues with the source card (hence why I want to clone it), its a bit unstable and slow on my Raspberry Pi. However I've managed to store it as an img file with `dd` so its not completely broken I'd say. (however when cloning the img file to another SD card using Etcher or dd the card isn't bootable on the RPi... which is why I bought the SD card reader to be able to clone it directly.)

martinkrulltott, 2021-12-29:
I tried shifting the cards in the card readers, now Im getting this error instead (note its also at 44% like last time) `ETIMEDOUT: connection timed out, read`. The log says the following:

```
gui.js:35 Flashing 1 device, 44% at 24.00 MB/s (total 24.00 MB/s) eta in 740s with 0 failed devices
2gui.js:35 Flashing 1 device, 44% at 22.50 MB/s (total 22.50 MB/s) eta in 789s with 0 failed devices
gui.js:46 Wed Dec 29 2021 21:32:50 GMT+0100 (Central European Standard Time) Write failed ({"image":{"path":"/dev/disk6","displayName":"/dev/disk6","description":"/dev/disk6","size":31914983424,"drive":{"size":31914983424,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk6","error":null,"isReadOnly":false,"displayName":"/dev/disk6","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk6","busVersion":null,"isSystem":false,"busType":"Secure Digital","isCard":true,"isUSB":false,"devicePath":"IODeviceTree:/arm-io/apcie@90000000/pci-bridge1@1/pcie-sdreader@0","mountpoints":[{"path":"/Volumes/boot","label":"boot"}],"description":"Apple SDXC Reader Media","isUAS":null,"partitionTableType":"mbr"}},"drives":[{"size":31914983424,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk4","error":null,"isReadOnly":false,"displayName":"/dev/disk4","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk4","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/arm-io/usb-drd2@2280000/usb-drd2-port-hs@02100000","mountpoints":[],"description":"Generic MassStorageClass Media","isUAS":null,"partitionTableType":"gpt"}],"driveCount":1,"uuid":"e8fd5fe2-d4ef-41be-aadc-3673f1a3d971","status":"failed","flashInstanceUuid":"e8fd5fe2-d4ef-41be-aadc-3673f1a3d971","error":{"name":"Error","code":"ETIMEDOUT","syscall":"read","errno":-60},"sample":0.1,"applicationSessionUuid":"7e43f168-92b5-4365-a6a2-428352aad653","flashingWorkflowUuid":"8102f677-3c81-431f-a740-af269e7bc4b0"})
gui.js:46 {"stack":"Error: ETIMEDOUT: connection timed out, read","message":"ETIMEDOUT: connection timed out, read","name":"Error","code":"ETIMEDOUT","syscall":"read","errno":-60,"image":"disk6"}
console.<computed> @ gui.js:46
gui.js:46 Terminating IPC server
gui.js:46 Flash results Object
o95242.ingest.sentry.io/api/6075635/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=d7bc1f9e943b463f8ddf3c5ec437e50d:1 Failed to load resource: net::ERR_CERT_INVALID
```

matthewmarcus, 2023-01-04:
Hey @martinkrulltott. Did you ever find a solution to this issue? I'm having the exact same problem trying to clone an SD card using balena etcher v1.13.1 on my Mac.

mcraa, 2023-01-04:
Not we did not.
It is difficult to reproduce without the same broken(?) sd card.
By the limited info we have here, the best guess is that the source of the issue is the card itself.
`dd` is working but the result is not bootable means it is not really working. dd is a "dumb" tool if it finds a broken byte it copies a broken byte.
Etcher freaks out as you see, to fix/recover that data is not a small project.

lurch, 2023-01-04:
AIUI, there's two "levels" of corruption on SD cards - at the filesystem layer and at the hardware layer.
Corruption at the filesystem layer can easily happen if you do an "unsafe shutdown", because the OS might be writing to a file at the same time as the power to the card is removed, which can lead to inconsistent filesystem data. Depending on which file(s) got corrupted, this might go unnoticed, and would only lead to "invalid" data being read when you next try to open that particular file. Or _if_ it's a more-important file, it might cause the whole OS to become unbootable. But filesytem corruption wouldn't (shouldn't?) stop Etcher from creating a backup of that SD card. And the card could still be reformatted / restored from a backup / etc.

But if you're unlucky enough to get hardware corruption (which touch-wood hasn't happened to me yet) then there's part of the memory in the SD card that has become "worn out", and can't be read or written, and in this situation Etcher _wouldn't_ be able to create a backup of the card, and trying to reformat the whole card or write a fresh image to the card would probably also fail. And again, depending on where the corruption happens, this may or may not prevent the OS from booting. As @mcraa mentions, `dd` is a "dumb" tool and will happily copy / ignore any broken blocks, which may not be desirable behaviour, as it can lead to a false sense of security! But if you're desperate to still recover data from such a "broken" card, you might want to have a look at https://www.cgsecurity.org/wiki/TestDisk

Of course this is all IMHO, and may not be entirely accurate.

matthewmarcus, 2023-01-04:
Thank you all for your responses. Much appreciated.

Yes, I concluded also through research and communication in various groups that the card itself probably has a hardware failure. As such, I'm gonna either recreate from a rather old backup or just rebuild what was on it.

Thankfully, the card still boots and will allow me to hopefully copy some files from it onto an external drive.

I'll also give the TestDisk a try just in case it might be helpful.

PaulBartelings, 2023-06-03:
Format the card or drive in fat32 will do the trick :)

closed 2023-07-12 by dfunckt