balena-io/etcher #3671: New Release Tempfile Issue

meawoppl, 2021-12-29:
- **Etcher version:** 
1.7.3

- **Operating system and architecture:** 
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

- **Image flashed:** 
Image from balena-cloud:
balena-cloud-TeleTom-raspberrypi0-2w-64-2.87.16+rev1-v12.11.0.img

- **Do you see any meaningful error information in the DevTools?** 

[image]

[image]

```
Wed Dec 29 2021 14:34:12 GMT-0800 (Pacific Standard Time) Write failed ({"image":{"size":952107008,"compressedSize":151123521,"hasMBR":true,"path":"/home/meawoppl/Downloads/balena-cloud-TeleTom-raspberrypi0-2w-64-2.87.16+rev1-v12.11.0.img.zip","partitions":[{"offset":4194304,"size":41943040,"type":12,"index":1},{"offset":46137344,"size":335544320,"type":131,"index":2},{"offset":381681664,"size":335544320,"type":131,"index":3},{"offset":717225984,"size":234881024,"type":15,"index":4}],"isCompressed":true,"name":"balena-cloud-TeleTom-raspberrypi0-2w-64-2.87.16+rev1-v12.11.0.img","extension":"zip"},"drives":[{"size":8169455616,"isVirtual":false,"enumerator":"lsblk:json","logicalBlockSize":512,"raw":"/dev/sde","error":null,"isReadOnly":false,"displayName":"/dev/sde","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/sde","busVersion":null,"isSystem":false,"busType":"USB","isCard":null,"isUSB":true,"devicePath":"/dev/disk/by-path/pci-0000:02:00.0-usb-0:2.3:1.0-scsi-0:0:0:0","mountpoints":[],"description":"SD_Card_Reader","isUAS":null,"partitionTableType":null}],"driveCount":1,"uuid":"c5606899-6b36-44aa-91a0-d28078800978","status":"failed","flashInstanceUuid":"c5606899-6b36-44aa-91a0-d28078800978","error":{"errno":-13,"code":"EACCES","syscall":"open","path":"/tmp/etcher/balena-etcher-electron-c9e5483390af.cmd"},"sample":0.1,"applicationSessionUuid":"b3fc1ebd-3011-4749-989c-133274c15d95","flashingWorkflowUuid":"702f9b63-3ee7-4ad1-b6f8-bf61dd3ca491"})
gui.js:46 {"stack":"Error: EACCES: permission denied, open '/tmp/etcher/balena-etcher-electron-c9e5483390af.cmd'","message":"EACCES: permission denied, open '/tmp/etcher/balena-etcher-electron-c9e5483390af.cmd'","errno":-13,"code":"EACCES","syscall":"open","path":"balena-etcher-electron-c9e5483390af.cmd","image":"balena-cloud-TeleTom-raspberrypi0-2w-64-2.87.16+rev1-v12.11.0.img.zip"}
console.<computed> @ gui.js:46
gui.js:35  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
```

It seems like it is unable to open/access /tmp for some reason? I am used to it popping up an authentication dilogue when the flash starts. (It didn't which is possibly related?)

meawoppl, 2021-12-29:
Zip file is intact, and unzipping then flashing from the `.img` file fails in the same way.

lurch, 2021-12-30:
Perhaps the `/tmp/etcher/` directory is owned by `root`, and that's what's stopping your `meawoppl` user from being able to write `/tmp/etcher/balena-etcher-electron-c9e5483390af.cmd` ? :shrug: (just a random guess, I'm not an Etcher developer)

mcraa, 2021-12-30:
Looks similar to #3670
@meawoppl did you get the sudo prompt and did you provide the password?

meawoppl, 2021-12-30:
I did not get a `sudo` prompt, which I found surprising...

closed 2023-07-12 by dfunckt

wickywaka, 2023-11-03:
My /tmp/etcher directory was owned by root. I removed it and then the etcher gave me a sudo prompt when I tried to flash and the continued flashing