balena-io/etcher #3672: `https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease` repo giving `402  Payment Required` error

jdrch, 2022-01-01:
- **Etcher version:** N/A
- **Operating system and architecture:** Debian 11.2 AMD64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

Seems the repo is behind on payments(?):

```
# apt update
Err:18 https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease
  402  Payment Required [IP: 52.84.121.70 443]
Reading package lists... Done
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/debian/dists/bullseye/InRelease  402  Payment Required [IP: 52.84.121.70 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Any ideas?

cossali98, 2022-01-01:
Also me I have the same problem, but it seems related to the signature of repository because yesterday it worked without problems.
Maybe the signature is expired with the start of new year?

mwllgr, 2022-01-01:
Also have this problem. Might be similar to https://github.com/balena-io/etcher/issues/3511?

lurch, 2022-01-01:
Also similar to #3665

mcking230, 2022-01-02:
Same thing with opensuse repo

misouris, 2022-01-02:
Same thing with Debian 11 -  amd64

jmknoble, 2022-01-02:
Duplicate of #3665

rexut, 2022-01-02:
Same thing with Ubuntu 20.04 Focal (amd64)

ABCMoNa, 2022-01-03:
Thanks for developing Balena Etcherand providing it to the general public free of charge, and for also hosting an apt repository.
This said, please consider your options to stabilize this (apt repository) service - it would be much appreciated to see a repository which works for more than ~ six months.

antoxa78, 2022-01-03:
same issue in Linux Mint 20.02 package Repository doesn't work

jdrch, 2022-01-03:
> Thanks for developing Balena Etcher and providing it to the general public free of charge, and for also hosting an apt repository.
> 
> This said, please consider your options to stabilize this (apt repository) service - it would be much appreciated to see a repository which works for more than ~ six months. Thank you.

This is my 1st time having this issue, but considering I've never used Etcher on this particular machine, so I'm strongly considering just removing it and the repo.

Zorono, 2022-01-03:
The same here too.... Ubuntu 21.10 (AMD64)
```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/impish/InRelease  402  Payment Required [IP: 54.230.114.10 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu impish InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

iamjameswalters, 2022-01-04:
Happy to say the repo didn't throw any errors for me today. 👍

jdrch, 2022-01-05:
> Happy to say the repo didn't throw any errors for me today. 👍

I'll see what the `cron` email output for the job that relies on `# apt update` to complete without errors gives on my affected machine.

mcraa, 2022-01-05:
Thanks @ABCMoNa for the kind words, and everyone for reporting the issue.
As you may found out from similar issues, cloudsmith has a monthly limit of downloads what etcher reached in the current period.
We contacted them and they added 20% to our limit for now, we will work with them together to make your experience smoother in the future.

lurch, 2022-01-05:
That's great news @mcraa :+1: 
You can probably close #3511 and #3665 too?

xdpirate, 2022-01-09:
Why not just use Launchpad, like nearly every other software project uses for its PPA...? Still getting this error.

GlenCooper, 2022-01-09:
```
20220109T122432Z: crystamped@mab:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
20220109T122439Z: crystamped@mab:~$ sudo apt update | egrep ^E

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Err:11 https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu focal InRelease
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/focal/InRelease  402  Payment Required [IP: 65.8.249.91 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu focal InRelease' is no longer signed.
20220109T122528Z: crystamped@mab:~$ 
```

@mcraa Please provide a status update on this.  Thanks.

halfagascan, 2022-01-09:
so, whats the solution?sudo apt update
[sudo] password for me: 
Get:1 https://download.docker.com/linux/debian bullseye InRelease [43.3 kB]
Hit:2 http://deb.debian.org/debian bullseye InRelease                           
Get:3 http://security.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:4 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]                   
Err:5 https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease                                      
  402  Payment Required [IP: 13.226.97.127 443]
Reading package lists... Done
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/debian/dists/bullseye/InRelease  402  Payment Required [IP: 13.226.97.127 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

skipperTux, 2022-01-09:
Continue getting `HTTP 402 Payment Required`. With regards to #3501, is [Cloudsmith](https://cloudsmith.io/~balena/repos/etcher/packages/) a good option?
[Alternatives](https://alternativeto.net/software/cloudsmith-package/) that might be a better option: [Bytesafe](https://bytesafe.dev/) - free for public open source projects, [Open Build Service](https://openbuildservice.org/), [Launchpad PPA](https://launchpad.net/ubuntu/+ppas) and [Fedora Copr](https://copr.fedorainfracloud.org/), [Pulp](https://pulpproject.org/) - OSS and self hosted.

rschr, 2022-01-09:
Those 20% – https://github.com/balena-io/etcher/issues/3672#issuecomment-1005482209 – didn't last long, or so it would seem ;-) Looks like the interest in Balena Etcher's .deb packages keeps increasing :-)

(In addition to the suggestions for alternatives to Cloudsmith perhaps just the remark that a .deb repository can also – without exactly requiring rocket science – be hosted as a static website on any webspace, as I think it is the case for .rpm repositories, too; but of course something like that might already have been considered...)

mcraa, 2022-01-10:
@GlenCooper , @rschr figured it right.

Thanks for all the recommendations, we considered and still considering multiple solutions.

The downloads in this period now more than 2.5x of the last months.
The total downloads size of the last 3 days is close to the average of the past months.
However we are happy that etcher is getting more popular, but this sudden growth - even taking into account the new release -, is kind of suspicious.

lurch, 2022-01-10:
Maybe lots of people got a new SBC for Christmas? :rofl:

jdrch, 2022-01-10:
> The total downloads size of the last 3 days is close to the average of the past months.

I haven't checked my RSS feeds in a while, but it could be that some very popular blog or Reddit post recommended or linked to Etcher.

jdrch, 2022-01-10:
> Maybe lots of people got a new SBC for Christmas? 🤣

Perhaps, but I think fundaementally speaking a sudden increase in downloads should never be an issue for a repo.

rschr, 2022-01-10:
>  The total downloads size of the last 3 days is close to the average of the past months.

Exactly 3 days ago, there was a new Linux Mint point release (20.3 "Una"). And the interest in that new version was so massive that the project's servers went to their knees with data rates dropping to two-digit kbit-per-second figures or even connection refusals. Something I've never seen before with a new Mint release.

Coincidence? 

Usually people would have simply upgraded which is an exceptionally smooth procedure for a point release, but maybe a lot of downloaders did fresh installs and still a large part of them immediately installed etcher, too?

jprokopowich, 2022-01-10:
Still an issue for me.

Zorono, 2022-01-11:
just a kind suggestion... why don't you do a similar thing to https://pmateusz.github.io/linux/2017/06/30/linux-secure-apt-repository.html or just use launchpad and its alternatives

mcraa, 2022-01-11:
The download limit got doubled for now, hopefully it is enough at least for another 3 days.

closed 2022-01-13 by mcraa

mcraa, 2023-03-21:
thanks for the ping, but I don't have the power any more to act on this
cc @aethernet

jdrch, 2023-04-28:
@XMA2l [Fix](https://github.com/balena-io/etcher/issues/4034#issuecomment-1525156978).

lurch, 2023-05-03:
See the sticky issue: #4059

RubenVerster, 2023-06-12:
If you're here looking to delete this bloatware

```ls /etc/apt/sources.list.d```

Lists all the PPAs. Look for these:
balena-etcher.list
balena-etcher.list.save

`sudo rm -i /etc/apt/sources.list.d/balena-etcher.list.save`
`sudo rm -i /etc/apt/sources.list.d/balena-etcher.list`

The above should work with this update blocking from balena
But you can also run this to clean balena out
`
sudo apt purge balena-etcher
`

Can finally update my machine after a month :')