balena-io/etcher #3673: Support to flash Windows ISOs.

xTrayambak, 2022-01-03:
Etcher is a great tool, there's no doubt in that, however, I need to dual boot Win10 and Fedora, but I can't since Etcher can't make bootable Windows devices, it'd be great if it could do that, I've seen this feature be requested in the past, and it'd make stuff really easier if Etcher adds this function.

lurch, 2022-01-03:
Duplicate of #210

engineer22, 2022-01-03:
Based on how Etcher works (how it builds the filesystem), and how Microsoft Windows ISOs are supposed to decompress and build the filesystem, Etcher needs a lot of work to be able to flash Windows and Linux ISOs.

For now you can use WoeUSB via Linux, which is built just for this reason. It works flawlessly.

prorounak2005, 2022-01-16:
> Etcher is a great tool, there's no doubt in that, however, I need to dual boot Win10 and Fedora, but I can't since Etcher can't make bootable Windows devices, it'd be great if it could do that, I've [...]

Why not use the dd command to flash isos ? It works on all mac and linux systems and use rufus in windows to flash isos in windows. :man_shrugging:

Nmatt1, 2022-01-23:
Basically, Windows does not play well with how the install media is created. To be 100% sure that it'll work you will need to use the windows provided tools.

zwhitchcox, 2022-04-18:
This is a duplicate of #210

closed 2022-04-18 by zwhitchcox