balena-io/etcher #3674: Flash drive bricked

storjak, 2022-01-04:
- **Etcher version:** 1.7.3 Portable
- **Operating system and architecture:** Windows 10 Pro 19042.1415
- **Image flashed:** kubuntu-20.04.3-desktop-amd64.iso
- **Do you see any meaningful error information in the DevTools?**
Couldn't clean the drive, Command failed: diskpart /s C:\Users\xxxx\AppData\Local\Temp\etcher\tmp-ec541992060e.tmp
 (code 2147942405)

That's the error I get from the second time running this, it may have been different from the first time.

After I ran Etcher the first time, I gave me the flash failed error page.  Since then, windows will not recognize the disk.  If I try to open the disk it asks me to insert a disk into the usb drive.  Using DiskPart throws me the error "Cannot zero sectors on disk \\?\PhysicalDrive3. Error code: 5@0101000F".  Running CHKDSK tells me "The type of file system is RAW.  CHKDSK is not available for RAW drives."  Windows Disk Management just give me a variety of errors whenever I try to touch the drive with it.

So far, all my recovery options aren't working.  The drive was working until I tried to flash kubuntu onto it.  Maybe the drive actually just died or maybe etcher bricked it, I don't know.

lurch, 2022-01-04:
See #3669 ?

storjak, 2022-01-04:
> See #3669 ?

Yeah I went through the instructions on the page that post linked, and like I said, the win DiskPart utility throws an error when I try to use it with this drive.

storjak, 2022-01-04:
Also tried using `mkfs` on my linux laptop, gives me the error attribute "partition" not found

storjak, 2022-01-04:
Solved my issue.

Using [this guide](https://askubuntu.com/questions/144852/cant-format-my-usb-drive-i-have-already-tried-with-mkdosfs-and-gparted/933035#933035) I booted up linux, and used mkusb to restore the drive to a standard storage device.  Quite literally every single other tool I've used just refused to work, this was the only one.

The drive wasn't fully destroyed, as I thought, but I'm guessing something in the partition table was changed by the distro I was trying to install earlier.

Either way, it's a mess all around.

closed 2022-01-04 by storjak

devxzero, 2024-08-10:
I had the same problem. The VDS Basic Provider said via the Windows Event Viewer `Cannot zero sectors on disk \\?\PhysicalDrive11. Error code: 5@0101000F` when I tried Windows Disk Management.
`Diskpart` >`select disk 11` > `clean` gave `DiskPart has encountered an error: Access is denied.
See the System Event Log for more information.` even though I used admin permissions to run it.

Then I saw this post here about Ubuntu, but before installing Ubuntu on a good USB drive and then use mkfs, I thought: let's install Ubuntu on the bad drive. I installed it the disk using Rufus, which surprisingly worked. After that, without going to the Ubuntu setup, I cleaned the drive using Diskpart, which again suddenly worked.

So where the default Windows tools like Windows Disk Management and Diskpart aren't able to fix the disk, a tool like Rufus is.

Xevion, 2024-09-30:
> I had the same problem. The VDS Basic Provider said via the Windows Event Viewer `Cannot zero sectors on disk \\?\PhysicalDrive11. Error code: 5@0101000F` when I tried Windows Disk Management. [...]

Found this through Google, can confirm, I just managed to save my crappy 32GB PNY USB.

I went through (more or less) the exact same steps:

- Originally used Balena Etcher to install PopOS on my personal laptop, USB was flashed with MBR more than 2 years ago. Since then, done nothing. Sat in my room, dry more or less.
- Today, I identified the MBR record and tried to begin wiping the USB for usage once again, just as a physical backup.
- While my other PNY drive that only had a Windows ISO (repair attempt) on it was fine, disk management deleted volumes, wiped, and formatted it just fine, this MBR drive refused.
- DiskPart also failed, and the reported error in EventViewer is identical to what you mentioned (5@0101000F).
- Installed Rufus, selected "Non-bootable" without an image, formatted it, and suddenly the drive works - pops up in Explorer immediately.
- Then deleted the volumes, reformatted it like I originally intended, and it all works great. Drive is fixed!

Rufus, my beloved. ([link](https://rufus.ie/en/#download))

Jaymmander, 2025-02-04:
I had the same problem. And Rufus solved it with his magic. He takes my love. I just put any . iso of windows just so that it would pass the usb to nfts and after that I used belena and I was able to boot pop os! correctly without errors. Curiously, rufus also caused me the problem. It's cruel but fair.

n122vu, 2025-07-25:
Had the same issue on an 8GB USB drive.  Had used Balena Etcher to install Linux Mint.  Wanted to put a different distro on the drive but could not.  Nothing I tried in Windows 11 worked.  I found this thread, downloaded Rufus, and the drive is now useable again.  Uninstalled Balena.

evanchsa, 2026-04-22:
I too had a USB flash drive that could not be recognized by Windows but worked fine in Linux. I had used several different programs to write Live Linux distributions to the USB flash drive but found myself needing to share data between Linux and Windows.

In my case I did not want any of the existing data on the USB flash drive, so I used wipefs (wipefs -a /dev/sdX) to wipe all partition and filesystem signatures on the device. This step is potentially optional.

I then plugged the wiped disk in to the Windows machine and saw the same symptoms -- no diskpart clean, the drive could not be formatted, the system seemed confused. Rufus could sometimes find the device and format it as non-bootable but the behavior was unreliable at best and still did not give me a working drive even after formatting.

Finally, I unplugged the device and decided to remove the Device Manager entries for the device as I thought Windows was remembering too much information for the newly wiped disk.

You can do this by:

1. Open Device Manager
2. Click View > Show hidden devices
3. Expand the "Disk Drives" section
4. You will now see all USB flash disks or SD Cards whether connected or not
5. Right click on the Disk Drive that is giving you trouble and choose "Uninstall"
6. Plug the device in and format the drive successfully (hopefully)

There is also a tool from Microsoft: [Microsoft DevNodeClean](https://www.microsoft.com/en-us/download/details.aspx?id=42286) command line tool. It is used to clean up the orphan shadow volumes left behind by the USB devices that are not currently connected to the system. I did not need to use this myself but this was the next step in my troubleshooting.